KlipperScreen/KlipperScreen #1: Error when running install script

lanman1, 2020-09-16:
Fully patched Raspian on Raspi4 with Klipper and Mainsail installed from kiauh scripts.  When I run /scripts/KlipperScreen-install.sh It pulls down the packages then gives me this.

Collecting ruamel.yaml.clib>=0.1.2; platform_python_implementation == "CPython" and python_version < "3.9"
  Downloading https://www.piwheels.org/simple/ruamel-yaml-clib/ruamel.yaml.clib-0.2.2-cp37-cp37m-linux_armv7l.whl (550 kB)
     |████████████████████████████████| 550 kB 1.2 MB/s
Building wheels for collected packages: vext.gi
  Building wheel for vext.gi (setup.py) ... error
  ERROR: Command errored out with exit status 1:

....

 running install_scripts
  Install vext files:
  error: [Errno 2] No such file or directory: 'vext': 'vext'
  ----------------------------------------
  ERROR: Failed building wheel for vext.gi
  Running setup.py clean for vext.gi
Failed to build vext.gi

Any help you could give would be appreciated.

lanman1, 2020-09-16:
Full detail on messages.  If there are some log files that would help, please let me know where they are.

[New Text Document (3).txt](https://github.com/jordanruthe/KlipperScreen/files/5234828/New.Text.Document.3.txt)

jordanruthe, 2020-10-17:
Did you install the right packages?

I had the package installation part of the install script commented out by accident. Please let me know if the latest commit of the script changes the results.

jordanruthe, 2020-11-03:
The installation script should be working now. Unfortunately, there is an error in the installation of vext.gi that I haven't been able to track down/suppress, but it installs properly despite it.

closed 2020-11-14 by jordanruthe