KlipperScreen/KlipperScreen #2: Installation unable to take control of the screen

michelluccote, 2020-10-26:
I've installed KlipperScreen but the service log files indicates it is unable to take control of the screen. I have tried running it with an auto  log-in raspbian setup or with-out auto log in or with start up with no UI. In all cases it is unable to display. With no auto log-in the log file indicates:

Oct 26 10:55:02 raspberrypi xinit[766]: #011and start again.
Oct 26 10:55:02 raspberrypi xinit[766]: (EE)
Oct 26 10:55:02 raspberrypi xinit[766]: (EE)
Oct 26 10:55:02 raspberrypi xinit[766]: Please consult the The X.Org Foundation support
Oct 26 10:55:02 raspberrypi xinit[766]: #011 at http://wiki.x.org
Oct 26 10:55:02 raspberrypi xinit[766]:  for help.
Oct 26 10:55:02 raspberrypi xinit[766]: (EE)
Oct 26 10:55:17 raspberrypi xinit[766]: Invalid MIT-MAGIC-COOKIE-1 key/usr/bin/xinit: giving up
Oct 26 10:55:17 raspberrypi xinit[766]: /usr/bin/xinit: unable to connect to X server: Resource temporarily unavailable
Oct 26 10:55:17 raspberrypi xinit[766]: /usr/bin/xinit: server error
Oct 26 10:55:17 raspberrypi systemd[1]: KlipperScreen.service: Main process exited, code=exited, status=1/FAILURE
Oct 26 10:55:17 raspberrypi systemd[1]: KlipperScreen.service: Failed with result 'exit-code'.
Oct 26 10:55:18 raspberrypi systemd[1]: KlipperScreen.service: Service RestartSec=1s expired, scheduling restart.
Oct 26 10:55:18 raspberrypi systemd[1]: KlipperScreen.service: Scheduled restart job, restart counter is at 14.
Oct 26 10:55:18 raspberrypi systemd[1]: Stopped KlipperScreen.
Oct 26 10:55:18 raspberrypi systemd[1]: Started KlipperScreen.
Oct 26 10:55:18 raspberrypi xinit[768]: (EE)
Oct 26 10:55:18 raspberrypi xinit[768]: Fatal server error:
Oct 26 10:55:18 raspberrypi xinit[768]: (EE) Server is already active for display 0
Oct 26 10:55:18 raspberrypi xinit[768]: #011If this server is no longer running, remove /tmp/.X0-lock

jordanruthe, 2020-10-26:
Which image of raspbian are you using? If something is already bound to display 0, then Xorg is starting up. After it boots up and fails, can you run `ps aux | grep Xorg` and see if there is anything for Xorg on :0 ?

If you ssh into the pi and run the following commands as sudo, you should be able to get it to start up.
`killall Xorg
systemctl restart KlipperScreen`

michelluccote, 2020-10-26:
I ran the ps command and found the Xorg running:

root       559  1.2  1.2 143280 50020 tty7     Ssl+ 10:51   1:37 /usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch

I ran the killall. Right now my Screen is showing the log-in window for Raspbian and it remains that way even after the two commands you sent me. Am I suppose to un-install the Raspbian UI to make Klipperscreen work?

michelluccote, 2020-10-26:
When doing the killall it appears to re-start it and the process appears again but with a different process id.

jordanruthe, 2020-10-26:
Try using raspi-config to make it so that it doesn't use a window manager and only shows the console. You can do this by selecting...
3. Boot Options
B1. Desktop/CLI
B1. Console

You probably need to reboot after these changes. That should disable the startup X session. I don't recommend using autologin, but that's up to you. It's not required for KlipperScreen

michelluccote, 2020-10-26:
It now show the console with command line login prompt. I checked the log files and now have a different error:

Build Date: 27 August 2020  08:51:48AM
Oct 26 14:10:59 raspberrypi xinit[695]: xorg-server 2:1.20.4-1+rpt2+deb10u1 (https://www.debian.org/support)
Oct 26 14:10:59 raspberrypi xinit[695]: Current version of pixman: 0.36.0
Oct 26 14:10:59 raspberrypi xinit[695]: #011Before reporting problems, check http://wiki.x.org
Oct 26 14:10:59 raspberrypi xinit[695]: #011to make sure that you have the latest version.
Oct 26 14:10:59 raspberrypi xinit[695]: Markers: (--) probed, (**) from config file, (==) default setting,
Oct 26 14:10:59 raspberrypi xinit[695]: #011(++) from command line, (!!) notice, (II) informational,
Oct 26 14:10:59 raspberrypi xinit[695]: #011(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Oct 26 14:10:59 raspberrypi xinit[695]: (==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Mon Oct 26 14:10:59 2020
Oct 26 14:10:59 raspberrypi xinit[695]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Oct 26 14:10:59 raspberrypi xinit[695]: (EE)
Oct 26 14:10:59 raspberrypi xinit[695]: Fatal server error:
Oct 26 14:10:59 raspberrypi xinit[695]: (EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
Oct 26 14:10:59 raspberrypi xinit[695]: (EE)
Oct 26 14:10:59 raspberrypi xinit[695]: (EE)
Oct 26 14:10:59 raspberrypi xinit[695]: Please consult the The X.Org Foundation support
Oct 26 14:10:59 raspberrypi xinit[695]: #011 at http://wiki.x.org
Oct 26 14:10:59 raspberrypi xinit[695]:  for help.
Oct 26 14:10:59 raspberrypi xinit[695]: (EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
Oct 26 14:10:59 raspberrypi xinit[695]: (EE)
Oct 26 14:10:59 raspberrypi xinit[695]: (EE) Server terminated with error (1). Closing log file.

The Xorg.o.log file now shows this:
[   170.131]    ABI class: X.Org Video Driver, version 24.0
[   170.131] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   170.131] (II) FBDEV: driver for framebuffer: fbdev
[   170.131] (EE)
Fatal server error:
[   170.131] (EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
[   170.132] (EE)
[   170.132] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[   170.132] (EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
[   170.133] (EE)
[   170.133] (WW) xf86CloseConsole: KDSETMODE failed: Bad file descriptor
[   170.134] (WW) xf86CloseConsole: VT_GETMODE failed: Bad file descriptor
[   170.134] (EE) Server terminated with error (1). Closing log file.

michelluccote, 2020-10-26:
I found a recommendation to give access to tty to my user is pi. To do this I put in the command:

sudo gpasswd -a pi tty

This changed my error and now the error is:

Oct 26 15:24:55 raspberrypi xinit[1466]: (EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)

From there I found more info and did this command:

sudo apt-get install xserver-xorg-legacy

Edited my /etc/X11/Xwrapper.config and but in the line:

needs_root_rights=yes
allowed_users=anybody

and it finally started to run with-out rebooting. So I don't know why by default it does not work. Maybe the setup script needs to take my scenario into account because it is painful to say the least to get it running even if I downloaded a default raspbien 32bit version which I think is the current latest that I used. 

If you need more info on my setup to better understand this then please ask and I'll be happy to provide any help I can.

Thanks

mhackney, 2020-11-02:
@michelluccote I was having very similar issues and came across your thread here. Your last post doing these things:
sudo gpasswd -a pi tty
sudo apt-get install xserver-xorg-legacy

editing /etc/X11/Xwrapper.config

also worked for me without a reboot. KlipperScreen started right up but is stuck on a page with the Klipper logo and "Connecting to Moonraker" and button to restart and restart firmware. The buttons do not function. I have had Moonraker installed and working with mainsail and fluid. I'll try restarting the pi and see if that helps.

mhackney, 2020-11-02:
reboot did not help and I had to run systemctl restart KlipperScreen to start KlipperScreen. My Moonraker.log file is clean up to the last lines with:
```
2020-11-02 08:22:00,908 [web.py:log_exception()] - 401 GET /printer/info (127.0.0.1): Unauthorized
2020-11-02 08:22:00,917 [web.py:log_request()] - 401 GET /printer/info (127.0.0.1) 10.40ms
```

But I don't know if this is from fluidd. However, the fluidd interface is working properly to control the printer but KlipperScreen is frozen on the restart screen I mention above. My Moonraker.log is attached.
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5475320/moonraker.log)

mhackney, 2020-11-02:
Ok, the error provides the key to the solution. You need to add 127.0.0.1 to your moonraker.conf trusted clients list. 

However, I still needed to run systemctl restart KlipperScreen but presumably when I reboot I won't have to.

jordanruthe, 2020-11-02:
Appreciate the feedback. I'll be looking at installation from scratch on rasbian again and make sure the installation script will take into account everything you both have said above, with the exception of the moonraker trusted clients configuration. I'll mention that in the installation notes.

michelluccote, 2020-11-02:
> Ok, the error provides the key to the solution. You need to add 127.0.0.1 to your moonraker.conf trusted clients list.
> 
> However, I still needed to run systemctl restart KlipperScreen but presumably when I reboot I won't have to.

I has a similar issue and it comes back everytime I connect with a different device. This is because the authorization file does not contain the info for the new machine to authorize it. I tried the instructions that say how to authorize any device on your current network and it never worked. I tried to put the current ip address and that did not work either. 

What did work is I look at the moonraker log file and it tells me the hardware address for the machine that is is not authorized. This is a set of hex digit separated by : characters. I then put that in the moonraker config file as a new line then stop and restart moonraker and it works. It sees the machine hardware address and sees it in the authorization list and it authorizes it. I have 3 different devices I use and each one is an item in the file. I would prefer the local network single line but like I said it never worked for me.

jordanruthe, 2020-11-03:
I just did a fresh install of raspbian and updated the installation script. I didn't experience needing to add pi to the tty users group, the main change I had to do for it to work on a fresh install was changed the Xwrapper.config to all any user, not just console users. 

The installation script should be working now from a fresh install.

closed 2020-11-11 by jordanruthe