KlipperScreen/KlipperScreen #3: Many of the buttons are not functional

michelluccote, 2020-10-26:
I am able to operate most of the buttons found under Actions but almost everything else does not work. Here is a description of the behavior of my install

1. The bed level to manually move the nozzle to each corner for manual bed level does not do anything. The Home and disable buttons also do not do anything in this menu.
2. The Z-offset menu allows me to move the nozzle and adjust it. I can also press the set z-offset button. Even if these operate the displayed z-offset remain 0 at all time. If I save exit the menu and enter it again it places the nozzle back a few mm off the bed exactly like before. One more detail. After I went back into this menu nothing works anymore. I went out and did home again and then back in and again nothing works anymore. I'll have to do a restart instead.
3. I can set the nozzle heat in the Actions preheat sub-menu but I cannot do anything in the top level preheat menu where I select between PLA and ABS. 

Most of the simple move X, Y and Z seem to work. I can control the fan and fan speed and do home from the Actions->Move menu but the rest is not functional.

I did look at my /var/log/syslog and for the operations that don't work I don't get any lines but the ones that do work I get these line:

Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "server.files.list", "params": {}, "id": 2}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "G28 X"}, "id": 3}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "G28"}, "id": 4}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "PROBE_CALIBRATE"}, "id": 5}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "TESTZ Z=-1"}, "id": 11}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "TESTZ Z=-.01"}, "id": 15}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "ACCEPT"}, "id": 16}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "SAVE_CONFIG"}, "id": 17}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "ACCEPT"}, "id": 18}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "SAVE_CONFIG"}, "id": 19}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "ABORT"}, "id": 20}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "G28"}, "id": 21}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "G91\nG1 Z1"}, "id": 25}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "M106 S0"}, "id": 35}
Oct 26 15:40:55 raspberrypi xinit[1721]: {"jsonrpc": "2.0", "method": "printer.gcode.script", "params": {"script": "M104 T0 S130.0"}, "id": 57}

michelluccote, 2020-10-26:
I am looking at mainsail console to see what commands are seen. 

I entered the z-offset window and I see not command echoed to the console. I'm assuming it should be seeing a PROBE_CALIBRATE. I then select to move the nozzle down and get:

10/26/2020, 4:03:16 PM | // Unknown command:"TESTZ"

Saving of course give the same Unknown command but for "ACCEPT" instead. Existing the menu returns a Klipper is ready.

Clicking Preheat ABS does nothing. Which configuration will define the temps for these options?

Most of my options not working appear to not be sending any commands to the printer.

jordanruthe, 2020-11-02:
I need to work on the manual bed leveling page, I think you're accurate that the buttons don't do anything currently.

The auto calibrate should be working, but there is an error case I need to spend more time on, which is aborting before it has finished the probe_calibrate command. I'll spend some more time looking into this.

For the preheat, it may be slightly unintuitive currently. You have to select the tool/bed first (so it is highlighted), and then hit the preheat option. If you have comments on how to make this better, let me know.

The temps are configured in the main directory, KlipperScreen.config file. I'll update the documentation to point this out.

michelluccote, 2020-11-02:
Just a quick note. I looked at your source code and see that the z offset command needed to start is part of the initialization function for that UI view. I actually had it work once and then didn't work the second time. I didn't have enough time to dig in more but I suspect this will work the first time you go into this bed leveling view but it won't after because initialization is probably only called ones.  It really should start every time you hit the z offset button in the parent UI of this panel and not in the initialization. Might be wrong but thought it might be worth provided this info for you in case it helps.

jordanruthe, 2020-11-03:
You're correct in noticing that the calibrate command will only run once. I'll re-hash this in the near future.

jordanruthe, 2020-11-04:
For the manual bed level, the buttons should be working now. I added the emergency stop just in case. 

The locations are configured from the bed_screws configuration section, or calculated if it isn't in the klipper config. 

For the probe_calibrate, it should run the calibrate command everytime you enter the menu now.

michelluccote, 2020-11-04:
Excellent. I'll updated my install and try it again when I finish my current long prints. 

One important request for me that is critical for adoption. When a print starts I really need a few buttons to adjust my z-offset on the fly. Bed leveling is never perfect and a few adjustments at the beginning of a print usually saves me from restarting. Any chance that could be added? Maybe +0.01, +0.05, -0.01, -0.05 buttons? These are the settings used by the Mainsail UI when it starts.

jordanruthe, 2020-11-05:
From the printing status screen, if you hit Control and then hit Tuning, there are buttons fine tuning z stepping, fan, speed, and extrusion factor. Currently these buttons aren't working, but I'll put this next on the list to finish that panel and get it working. I might have to break z baby-stepping into it's own panel as it's getting pretty busy on that panel.

jordanruthe, 2020-11-14:
The fine tuning panel (including z babystepping) should now be fully functional

michelluccote, 2020-11-15:
That is excellent news. I will be trying it this week. I have been using the mainsail page directly but find it not ideal for a touch screen since I don't have a keyboard connected. If this is all working it will a better way to control when at the printer. Thank you for all the great support.

closed 2020-11-16 by jordanruthe