KlipperScreen/KlipperScreen #4: the print screen does not display the newly recorded gcode

vacapetr, 2020-10-30:
hi, the print screen does not show the newly uploaded gcode. The restore button on the screen does not work. The service must be restarted "sudo systemctl restart KlipperScreen.service". Additionally, the uploaded file does not match. it shows me 50 years ago.

jordanruthe, 2020-11-03:
The print screen should be updating now. Currently, directories are not working properly, I'll be addressing that in the future.

For the file time, I removed a function that modifies it. If it is still showing an incorrect time, something is wrong with the time on your system, or it's improperly reporting from moonraker.

vacapetr, 2020-11-09:
Hello, the newly uploaded g-code will already be displayed.
The time is still displayed incorrectly. Mainsail displays it correctly.
[image]
[image]

jordanruthe, 2020-11-09:
The time should now be displayed in the format of 2020/11/09 16:38

vacapetr, 2020-11-10:
I confirm the time ok

closed 2020-11-11 by jordanruthe