KlipperScreen/KlipperScreen #5: Screen goes black and won't wake

mhackney, 2020-11-02:
7" HDMI panel. There is no menu on it to set screen saver time so this must be happening external to it. After just a few minutes the KlipperScreen goes black and won't redisplay until I turn the panel off, on and then press on again. The first time I turn on I see the RPi log and then an "Input not Found" error, the second On press activates KlipperScreen. Is there a config somewhere to completely disable this time out?

mhackney, 2020-11-02:
Ok, with some digging I did find a disablescreenblank.sh into scripts directory. Maybe a little documentation would help cut down the wasted time and posts? I'm happy to help with docs if time is the issue.

jordanruthe, 2020-11-02:
The disablescreenblank.sh should have been running from the startup script. I just pushed a change to include that, but you'll have to re-run the installation script.

mhackney, 2020-11-02:
great, thanks

jordanruthe, 2020-11-02:
Sure. Also, feel free to contribute any documentation you would like.

closed 2020-11-02 by jordanruthe