KlipperScreen/KlipperScreen #17: CantKlipperScreen.service: Failed to execute command: No such file or directory

michelluccote, 2020-11-15:
I updated my KlipperScreen by downloading the latest version using a clone command. Then ran the installation script which ended with the following message:

Updating X11 configuration
Job for KlipperScreen.service failed because the control process exited with error code.
See "systemctl status KlipperScreen.service" and "journalctl -xe" for details.

When I run systemctl I get this message

jordanruthe, 2020-11-15:
Just pushed a fix, I had committed the installation script with the install packages line commented out.

michelluccote, 2020-11-15:
I re-installed it and it worked on my first install script run. Thanks for the very quick fix. Now to test it. Any functionality not yet implemented in the UI?

michelluccote, 2020-11-15:
First thing I did was the z-offset calibration. The operation seems to work but the diplayed z-offset while I'm adjusting just shows the same number through out the process. Mine was showing -8.0. Looking at the mainsail UI I can see the real values and they are being adjusted. The display should be adjusted.

Once I accept I see that Klipper exits and is restarted. This is intended? 

If I go back to z-offset menu it still shows 8 even if mainsail tells me it's 6.270. This number seems off as well and is shown by the nozzle being way off the plate when I go back into the z-offset menu. If I do a G1 Z0 it does go back to the right z-offset. Is this expected behavior? Every time we go into z-offset it will start back way off the plate? Maybe that is correct to avoid collisions so it might be fine but the displayed offset still needs to be adjusted.

I then went into the bed leveling UI and it displays nothing. Just a blank screen. Where do I go in the system to see if an error happened? When this happens I am stuck doing a restart since there is no button even to remove to previous screen.

This is one more point. Klipper and moonraker have log files that are either in tmp or can be moved to a good location in my pi account. Can KlipperScreen do the same so I can inspect it in case of an error?

michelluccote, 2020-11-15:
I noticed when I start a print the screen resolution changes and my view is clipped on the left side of the screen? I am using the original 7" display and it works find up until I start a print.

michelluccote, 2020-11-15:
tried printing but the z-offset adjustment during printing is displaying 0 and when I press up or down it remains at 0 and the nozzle does not move. I looked at mainsail and no command is received to move the nozzle.

michelluccote, 2020-11-15:
Could not start a print. My file does a G29. After ABL the nozzle is pushed agains the bed for a reason I don't understand. Need to do a 2mm or so adjustment up to fix this but the UI does not allow me to make any changes. While printing my computer connections got very slow so  I could not make this adjustment using mainsail. It even disconnected. Had to do a emergency shut down. After I did a homing the buttons work again but now I don't know what sequence will allow me to print again. I also noticed there is no place I can confirm exactly which version of KlipperScreen I'm using to make sure it is the latest.

jordanruthe, 2020-11-16:
Thanks for the feedback. At this point, there should be no buttons displayed that are not implemented.

Let me start by first saying that there is a log file. It's located at /tmp/KlipperScreen.log
I'll be pushing an update shortly to increase the logging of the log file so I can see exceptions and version number. The version number is also displayed now in the System Information panel.

For the z-offsets, there may have been an issue with parsing the gcode response, but it should be working now (I just tested it on my own machine). There was a problem with the fine tuning z offset as well, and I had to implement a fix for it to work as expected.

I haven't ran into any issues like you've described with printing, and I've even had it do a bed mesh update. I'd be interested in seeing the output of your log file if you run that scenario again.

jordanruthe, 2020-11-16:
I'm assuming the G29 macro you have runs BED_MESH_CALIBRATE. Are you homing the Z axis again after the BED_MESH_CALIBRATE command?

michelluccote, 2020-11-16:
No I am not doing a G28 after my G29. So my gcode would need G28; G29; G28. I've not seen that requirement before. I'll check my setup but I can always redo a G28 as part of my macro if that is needed. I'll test it to see if already homes or if it goes straight to printing. My current print should be done in a few hours and I can test it as soon as its done.
Thank you for the great support and help.

michelluccote, 2020-11-16:
For the blank screen are you saying there is no functionality yet available in the bed leveling menu? If that is the case there should at least be a return to previous screen option. For now it is blank with no option and you are force to stop and restart KlipperScreen to get control again.

jordanruthe, 2020-11-16:
Depending on what your macro is doing, a G28 Z might be required. If you load a bed_mesh_profile, you have to re-home. I've also ran into issues before with calibration, so I always re-home the Z axis after doing anything with ABL. 

For the blank screen, there is an error that is preventing the panel from properly loading. If you're on the latest version, the log should be able to tell me what the error is so that I can correct it. I'll also be implementing a fall back feature shortly so that it won't blank screen in the future, it will gracefully fail.

michelluccote, 2020-11-16:
Thank you for this info. I did a quick test and adding the G28 did resolve the issue. I looked at my klipper macro and it does save the mesh and reload it and it does not do a Z-axis re-home. I'll do a tweak to it and this problem should be gone. I've struggled with this for such a long time so I'm very appreciative of your help on this problem.

jordanruthe, 2020-11-17:
The easiest way to update would just be to run a "git pull" on KlipperScreen. It should pull the latest and update the files for you along with showing what files have changed.

You don't need to keep running the install script. I'll write a doc in the near future for updating to detail all this.

michelluccote, 2020-11-17:
I updated the KlipperScreen package to the latest version. Ran the install script and rebooted. The version is f223f74 shown in the system info. When I go to bed leveling it now gives me an error indicating to go to the /tmp/KlipperScreen.log file. There is an error in this file and I've attached the file to this message. 
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5551175/KlipperScreen.log)

michelluccote, 2020-11-17:
After doing this test I decided to redo my z-offset. The printer seemed to do the homing and came back to get ready for the zoffset. It went down hard on the bed and kept going. Heard something crack before I got the power off. No idea why it did that. After a restart is worked again but some sequence of even caused this to happen.

For the blank screen I got it one more time. My sequence was to go into z-offset calibration. Make the adjustments. Usually when I press accept it restarts klipper screen and goes to the main screen. This time is said nothing but went one level up in the screen. I notice in mainsail it did not send the accept message. I then pressed bed level and it went blank. Had to stop and restart the KlipperScreen service. 

Here is the log saved just after that happened.
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5551289/KlipperScreen.log)

michelluccote, 2020-11-17:
I did notice many operation can't happened if the machine was not homed. Is that controlled through the macros or the UI could indicate a home is need and ofter the operation before continuing?

michelluccote, 2020-11-17:
I still have a lot of the same problems. First my start up screen has the correct resolution but after a few menu selections it scales and trips the right side. Here is a picture:
[image]

Then I start a print after doing z-offset and the nozzle is too far off the bed. I do the small adjustments which I can see in the KlipperScreen.log but not in the moonraker.log. I also don't see it going through in the mainsail web page. I'm including both files save right after I cancelled the job.
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5551340/KlipperScreen.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5551341/moonraker.log)

michelluccote, 2020-11-17:
To clarify the z-offset adjustment during printing did not work in the description above. 
One other detail missing is when I go to the bed level view the initial z-offset it display is some very large number like 643. Just as it starts adjusting the nozzle down it then switches to display my real 1.5mm value.

jordanruthe, 2020-11-17:
Thank you for all the feedback. It's really helpful! I'm sorry to hear about the head crashing into your bed though.

I'll be going through this when I get a little bit of time later. I can confirm that the z offset panel is not checking homes axis, so I will fix that and run a home if necessary. 

For the resolution, vacapetr has been submitting some pull requests to help with that. I'll try looking at the panels as well, but I've been doing this work on a 1024x600 resolution screen. I've been meaning to do ui scaling for smaller resolution and will start working on that.

michelluccote, 2020-11-17:
I was doing a few searches on the resolution scaling problem and found that octoscreen had the same issue. It appears to be due to the file name length causing the UI to stretch. The issue was resolved by limiting the file name length to 20 characters in their case before displaying it on the print UI. Maybe some other views will have a similar issue. I can test that later as well and see when this happens and if it is due to the same problem.

Thanks for looking at this and updating the code. I think this UI is going to be really great once some of these details are all fixed and the functionality is complete. For me adopting klipper is entirely dependent on having a functional screen on the PI. I find controlling the printer from my computer of phone is cumbersome because the response time is too slow. I may have to boost my network near the printer since my wifi is slow but it will never be as responsive as dealing with the PI directly.

michelluccote, 2020-11-18:
I can confirm that the file name length is not the problem with the screen scaling. I started a long file name no issue and UI was still ok. I went into the control menu and then find tune and voila it is already having the problem. Something about the fine scale UI is causing it.
To test this I did a bunch of operations and all menus are fine and seem to have no effect on the UI fitting in my screen. The one that causes the screen to stretch horizontally seems to only be caused by the control menu->fine tuning sub menu while you are printing. As soon as you enter that screen it is already displaying a stretched view and cutting the right edge off the UI like in my previous picture.

michelluccote, 2020-11-18:
Good news. I figured out the problem with the screen scaling. The string lengths on all your grid labels are too long. With higher res you can fit them in but on my screen having all these long labels side by side like "Increase Temperature", "Increase Speed" and "Increase Extrustion" causes the screen to stretch and it never recovers. I changed the "Increase" to "Incr" and "Decrease" to "Decr" and did a restart of KlipperScreen and it now work with-out reseting the UI width. Keeping labels short or using images will help avoiding this problem. 
My main remaining problem is having have the tuning work to adjust the z-offset during printing in this very panel.

michelluccote, 2020-11-18:
I experimented some more with the fine tune of the z. It looks like it is working. Initially the response seems a bit slow and I'm having a hard time adjusting it but after a few attempts it starts responding correctly. I've tracked the communication through moonraker and mainsail and all is in sync. So this seems to work.

I needed to adjust my z-offset by -0.28mm to get a good first layer. This tells me that probably my z-offset is off by that amount. The problem is that I have no way to go adjust my z-offset manually. I've done that in my SKR UI and I've done it in either the printer.cfg or using a mainsail command. I would love to go to z-adjust menu and just manually adjust my current value up or down and then save it for the future so I can get good starts with-out find tuning.

My other remaining item is the blank bed leveling panel. That would complete what I think is needed to get a functional KlipperScreen with-out needing to jump to mainsail. I'm sure other features would be great after but that would cover all the basics in my view.

jordanruthe, 2020-11-19:
The bed level panel should be working for you now. I attempt to get 4 screw locations from the bed_screws configuration section in klipper. If that is not there, KlipperScreen will automatically calculate 4 positions on the bed.

I don't know that having a manual z-offset adjust is a good way to go. This should be fixed if the probe calibration is done properly, and then you can save the config from the configuration menu.

michelluccote, 2020-11-20:
Excellent. I'll re-install. Did you get a chance to look at the long labels in the fine tune menu by any chance?

For z-offset I wish it was that easy for me. I adjusted my z-offset very carefully using a 0.1mm metal gauge. Even done carefully using a 7 by 7 ABL step I always must do a small adjustment at the start of each print. This adjustment tends to be consistent so after two prints to confirm I usually adjust my z-offset and then I don't need to do it again for most prints. I do this for all 3 of my printers each with a different UI on them. I find z-offset calibration is based on a feel of how the nozzle slides across the gauge when adjusting the printer and that is not a very scientific method and is prone to small errors. My SKR v2 does have this option and I find it very convenient to make the adjustment. For Klipper I have to instead modify my printer.cfg file. If you still don't want to add it then that is fine. I can still work with what the UI already offers and worst case I can go to the printer.cfg.

One question for you. I'm providing a lot of feedback. Do you prefer if I submit different reports for each specific item or we can continue with this one longer conversation?

Thanks again for the great support. I know this is spare time work and I appreciate the commitment. Any chance this component could be added to kiauh?

jordanruthe, 2020-11-24:
The fine tune menu should be updated with shorter labels. I'll have to think about how that can be accomplished from a UI perspective. 

It's better for me to have issues for different topics mainly because it makes sure that I won't miss anything. But at the end of the day, as long as I'm getting the feedback either way works. 

For KIAUH, it's a possibility. I'd need to see if the maintainer would even want it included first however.

michelluccote, 2020-11-24:
I did notice and can confirm the shorter menus. I am having a few remaining issues and I'll file separate items. I think I've already created a few and I have a few more. It just take time because I do run long prints in between debug sessions.

Thanks for the info on KIAYH. 

I can confirm bed leveling is working but I will submit something for a recommended change. The code assumes 1/4 of bed in from the bed size. My bed is large so this brings the nozzle too far in. Bed size is printed size and not actual physical size to the max locations are offset in further from the corners. Don't worry. I'll file a separate item for this.

jordanruthe, 2020-11-28:
KIAUH now has KlipperScreen as an option

michelluccote, 2020-11-28:
That is great news. You work fast. This will help people looking for a solution like I am. Good.

michelluccote, 2020-11-29:
I updated KlipperScreen/Klipper/Moonraker/Mainsail to the latest versions using Kiauh. Now klipper is dead with this error:

Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 188, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 242, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, reqclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 180, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed

I refreshed the printer to make sure it was ok and it didn't help.

michelluccote, 2020-11-29:
Ok. Got it working after fixing the baud rate. KlipperScreen is still dead even if everything else is working. Mainsail also works and can control and I can print but KlipperScreen is dead. Please see attached log file.

Here is my version information:

/=======================================================\ 
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/
/=======================================================\ 
|     ~~~~~~~~~~~~~~ [ Update Menu ] ~~~~~~~~~~~~~~     | 
|  https://www.klipper3d.org/Config_Changes.html        | 
\=======================================================/
/=======================================================\ 
|  0) [Enable] backups before updating                  | 
|-------------------------------------------------------|
|  a) [Update all]       |  Local Vers:  | Remote Vers: | 
|                        |               |              | 
|  Firmware:             |               |              | 
|  1) [Klipper]          |  63077484     | 63077484     | 
|                        |               |              | 
|  Webinterface:         |---------------|--------------| 
|  2) [DWC2-for-Klipper] |  --------     | --------     | 
|  3) [DWC2 Web UI]      |  --------     | 3.1.1        | 
|                        |---------------|--------------| 
|  4) [Moonraker]        |  d41c1c4c     | d41c1c4c     | 
|  5) [Mainsail]         |  0.3.0        | 0.3.0        | 
|  6) [Fluidd]           |  --------     | 1.0.0        | 
|                        |               |              | 
|  HDMI Screen:          |---------------|--------------| 
|  7) [KlipperScreen]    |  cc59ec68     | cc59ec68     | 
|-------------------------------------------------------|

[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5611804/moonraker.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5611802/KlipperScreen.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5611803/klippy.log)

michelluccote, 2020-11-29:
My cut and past version info did not work well so here it is in picture form:

[image]

jordanruthe, 2020-11-29:
I just pushed an update. It looks like you don't have power devices and I wasn't handling that case properly in the startup. Let me know if that fixes it fo ryou

michelluccote, 2020-11-29:
I updated and I'm up and running again. Thank you for the quick update.

michelluccote, 2020-11-29:
Most of these issues have been resolved. I am closing this one and will open new items when encountered instead.

closed 2020-11-29 by michelluccote