KlipperScreen/KlipperScreen #18: No option to run a G29 Auto Bed Level

michelluccote, 2020-11-22:
My printer supports auto bed leveling with a BLTouch. Klipper is configured to run this command. My flow is to run it once and then no run this command before every print to save time. I can get many print going.

For now KlipperScreen allows me to do Z-offset followed by a manual bed leveling. Once done I would want to update my ABL table and run a G29 but this option is not supported yet in KlipperScreen.

Please add support for ABL.

jordanruthe, 2020-11-23:
Thanks for this feedback. You can add a G29 button yourself by using the KlipperScreen.conf file. I just pushed some updates which detail the new configuration file and how to configure it.

That being said, I do think it would be cool to be able to view ABL profile and add/delete them from KlipperScreen. I'll leave this open as a reminder for that.

jordanruthe, 2020-12-04:
I just added a bed mesh panel. I still have some ideas on how to improve upon it. If you wouldn't mind, could you check it out and give me any feedback you may have?

I would like to add the ability to create a new mesh profile, but I need to implement a touchscreen keyboard before I'm able to do that.

michelluccote, 2020-12-04:
Excellent. Excited to have ABL added. I will give it try after my current print which should be by noon tomorrow.

michelluccote, 2020-12-04:
My print I had just started had to be restarted so I got to take a quick look. 

The interface is simple enough. I was a bit confused because I guess my setup now had a "default" and a "DEFAULT" bed level. It displays both. The "DEFAULT" has a save and up arrow button while the "default" had an update button which because a up arrow when I updated the mesh. Can you tell me what the up arrow is? It's to read back the mesh for the current environment? I'm including a picture of my view.

One improvement would be to tie the bed temperature into this. I accidentally did a full bed level (81 points in my case) forgetting that I did not get the temperature up first. I'm thinking either display the temp and provide button to a temp control panel and then wait to reach temp before starting or when doing an updated ask at what temp the bed should be and then wait for it to reach it before leveling. In my case I had to wait for it to finish, go back to adjust the temp. Wait.... Rerun bed mesh leveling.

As you already mentioned a way to provide the name.

Finally of course a way to visualize this information would be great. I'm sure that is a lot more work but displaying the bed level really helps to know if something is wrong. Not sure if this is really requires mainsail or other web interfaces can take care of that in most cases.

I'll provide more feedback as I use it more. For now it worked for me once I got the sequence of operations right. Great enhancement.

jordanruthe, 2020-12-05:
The up arrow is "Load". The SD card icon is save, and the refresh is calibrate_bed_mesh. I'm playing around with how to make it include load/save/etc in without it taking up too much space. 

For visualizing, I do have a way to visualize the data but I'm trying to optimize it. It's caused some issues running on a pi with KlipperScreen crashing.

I appreciate the feedback!

michelluccote, 2020-12-08:
Thanks. I'm sure you will continue to refine this but for now I think I can close this since I have the basic functionality I require to run my printer.

closed 2020-12-08 by michelluccote