KlipperScreen/KlipperScreen #20: Manual bed level UI nozzle placement sub-optimal for larger bed sizes

michelluccote, 2020-11-24:
I am using KlipperScreen on a CR10 Max. It has a 450mm by 450mm printable bed size. The bed is physically 470 by 470. When doing manual bed leveling the UI uses 1/4 the size of the printable bed size to decide on the nozzle placement. For my print this shifts the nozzle into the bed to location X=112.5mm Y=112.5mm. This is too far from the corner of the print bed. 

When then moving the the max X and Y location it seems to use printable bed size - 1/4(printabel bed size). This results in a location of X=337.5 Y=337.5 (I've not confirmed this from the logs yet) which seems to result in a location even further in. I'll confirm this exactly by looking at the log once my current print has completed but I know 1/4 is used in the source code and you probably know exactly how physical vs printable area is used in this calculation.

A better method would be to use a user specified offset either from the printer.cfg or the KlipperScreen.conf. It could then use the offset combined with the bed physical size to calculate the nozzle location. This would result in a more controllable and predictable behavior. 

For now I have manually modified my source to use 1/8 instead but I still have the impression the nozzle is shifted in further on the max side. I'll confirm and provide the log shortly.

jordanruthe, 2020-11-25:
You're correct in the calculation. I think there are some things I can do better in terms of getting it closer to the edge. I'll take a look at this, but please check out the bed_screws configuration in Klipper. If you have four "screw" locations defined there, the bed level panel should use those locations instead of calculated locations.

michelluccote, 2020-11-28:
I added a section like this in my printer.cfg. Restarted klipper and confirmed it was loaded. I see a difference in the nozzle placement but it does not seem to align with my settings. I them modified it again and it is not changing the location of the nozzle. I don't see much in the KlipperScreen.log file to tell me what values it extracted from my printer config file.

Here is the current content of my settings. 

[bed_screws]
screw1: 40,40
screw2: 410,40
screw3: 40,410
screw4: 410,410

I'll do more testing once this current print completed.

jordanruthe, 2020-11-28:
In the log you should see one of two things

"Bed screw locations" will show up if it's using your configured values.
"Calculated screw locations" will show up if it isn't.

michelluccote, 2020-12-01:
When I press a location the only log information I get is this:

2020-12-01 08:47:57,136 - KlipperScreen.KlippyFiles - DEBUG - Got metadata for C10X25cmby1cmRoundFlat.gcode
2020-12-01 08:47:58,361 - KlipperScreen.BedLevelPanel - DEBUG - Going to position: [13.0, 40.0]
2020-12-01 08:47:58,373 - KlipperScreen.KlipperWebsocket - DEBUG - Sending printer.gcode.script: G90
G1 Z7 F800

G1 X13.0 Y40.0 F3600

G1 Z.1 F300

2020-12-01 08:47:59,128 - KlipperScreen.KlipperWebsocket - DEBUG - Sending server.files.metadata: C10X25cmby1cmRoundFlat.gcode

jordanruthe, 2020-12-02:
This should be in the log when you first open the Bed level panel. If KlipperScreen has been running for a while and you've already been into the panel, you'll need to restart it for that message to appear

michelluccote, 2020-12-03:
I redid my testing and found the screw information. The nozzle is not quite where I expect but it seems it's a problem with how I'm measuring the screw location. This works and I am sure I can finish getting it tuned. Thanks for the great support as usual.

closed 2020-12-03 by michelluccote

jordanruthe, 2020-12-08:
I just realized what was probably happening here. My code was taking into account the bltouch/probe offset, so if you're running ABL it most likely put the probe over where you wanted the screws. 

At any rate, I just made another change to the bed_level panel that allows it to support the SCREWS_TILT_CALCULATE command. Documentation on it is here: https://github.com/jordanruthe/KlipperScreen/blob/master/docs/panels.md#bed-level
If you haven't used the screws_tilt_calculate command, it's a pretty nice helper for manually leveling a bed.