KlipperScreen/KlipperScreen #21: KlipperScreen does not recover from a save config

michelluccote, 2020-11-26:
After doing a save config in KlipperScreen it restarts Klipper but then gets stuck on the "klipper has shutdown" screen. Klipper is back up and I can see everything is fine in mainsail. The logs from moonraker and from klipper seem find and working. Only KlipperScreen is stuck and won't run correctly unless I stop the service and start it again.

I've included the klippy.log and KlipperScreen.log and a picture of the image I am stuck on. I can also put the moonraker.log file if it helps but it is 40M in size.

[image]
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5599876/KlipperScreen.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5599877/klippy.log)

jordanruthe, 2020-11-27:
Can you upload the moonraker log? I'm unable to tell what happened with those two logs unfortunately.

michelluccote, 2020-11-27:
This is a reproducible problem in my environment. I will wait for my latest print to complete and will do a fresh restart and clear all logs. Then I'll reproduce it which should give me the smallest log files possible. I can then upload these for you. Probably will happen tomorrow morning.

michelluccote, 2020-11-28:
I've cleared all logs. Restarted all services and did a G29 followed by a save configuration. KlipperScreen never returned correctly. Please find attached all 3 log files for Klipper, KlipperScreen and moon raker. Mainsail does recover by the way and is fully functional even if KlipperScreen is still down.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5611218/KlipperScreen.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5611219/klippy.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5611220/moonraker.log)

jordanruthe, 2020-11-28:
I'm unable to verify at this time. But my best guess is that you're using a version of moonraker before they added a "notify_klippy_ready" message that I use to find when the printer is ready. Would you be able to update moonraker? This is the exact commit it was added: 3d827d7513946c06680d69b910c3c8777584cad0

michelluccote, 2020-11-29:
It's a pretty recent version but that might be the case. I'll updated it after this print and confirm. Do you know which version added that message? My moon raker log says I'm using Git Version: v0.1.0-51-g413062f

michelluccote, 2020-11-29:
OK. Looks like there is a v0.2 and I'm still on v0.1. I'll update it tomorrow morning and if all goes well we can close this one.

michelluccote, 2020-11-29:
I have updated to latest version and I can confirm this is now fixed. My KlipperScreen recovers and returns to its main panel as expected.

closed 2020-11-29 by michelluccote