KlipperScreen/KlipperScreen #23: ERROR - Fatal error in main loop (module 'websocket' has no attribute 'WebSocketApp')

fex42, 2020-12-01:
After doing a completely fresh install of Raspberry Pi OS, Klipper, Moonraker, Fluidd and KlipperScreen (using KIAUH), I have everything working except KlipperScreen which throws the following error in /tmp/KlipperScreen.log:

```
2020-12-01 16:43:10,120 - KlipperScreen - ERROR - Fatal error in main loop
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 465, in <module>
    main()
  File "/home/pi/KlipperScreen/screen.py", line 457, in main
    win = KlipperScreen()
  File "/home/pi/KlipperScreen/screen.py", line 104, in __init__
    self._ws.connect()
  File "/home/pi/KlipperScreen/ks_includes/KlippyWebsocket.py", line 61, in connect
    self.ws = websocket.WebSocketApp(self.ws_url,
AttributeError: module 'websocket' has no attribute 'WebSocketApp'
```

JohannesBertens, 2020-12-01:
I went into the Python virtual environment, removed the 'websocket' and 'websocket-client' package and did a reinstall of the requirements:

```
pi@klippi:~ $ source ~/.KlipperScreen-env/bin/activate
(.KlipperScreen-env) pi@klippi:~ $ pip uninstall websocket
(.KlipperScreen-env) pi@klippi:~ $ pip uninstall websocket-client
(.KlipperScreen-env) pi@klippi:~ $ pip install -r ~/KlipperScreen/scripts/KlipperScreen-requirements.txt
```

fex42, 2020-12-01:
Sadly the fix of @JohannesBertens doesn't work for me. I tried it (and some variants) without success :-(

JohannesBertens, 2020-12-01:
Oh, I forgot to mention - also removed the line with just 'websocket' in the KlipperScreen-requirements.txt file before executing the above commands, mine now looks like this:

```
requests==2.25.0
vext==0.7.4
websocket-client==0.57.0
humanize==3.1.0
jinja2==2.11.2
```

fex42, 2020-12-01:
Ok, that worked. Had to remove the websocket line. Thank you!

jordanruthe, 2020-12-02:
Thanks for reporting this. I just updated the requirements file per the feedback.

closed 2020-12-02 by jordanruthe