KlipperScreen/KlipperScreen #25: Unable to cancel a print with-out pressing the pause button

michelluccote, 2020-12-03:
When I try and press the Cancel button it just keeps on printing. After trying multiple times I usually press the Pause button and then it finally stops but seems to be in pause but the UI returns to the main panel. 

In the attached log files I started a benchy print. Let it warm up. When it started I pressed cancel three times but it continued. I then pressed pause and it finally stop. It still had heat going on both bed and hot end which I had to stop manually. It looks like it was in pause but the UI was on the main panel where I could start a new print which does not feel like pause. 

One other note. I had a hard time getting the files off my PI because the PI seemed to be unresponsive. I had to restart the firmware (and nothing else) before it came back to life and I could get the log files. Looks like the system was waiting on a firmware response or something. I've attached all the log files.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5634002/KlipperScreen.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5634003/klippy.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5634004/moonraker.log)

jordanruthe, 2020-12-03:
I pushed an update to fix the cancel button. I'll need to take a further look at it switching to the main page when you only paused it.

michelluccote, 2020-12-04:
I have not done the update yet but I did retest the pause. Same behavior. Paused went to main UI. Printer was in pause with heat still being maintained. Lost access of raspberry PI through wifi until I used KlipperScreen to restart firmware. Here are logs for the simple flow that reproduced this problem.
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5640414/KlipperScreen.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5640415/klippy.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5640416/moonraker.log)

michelluccote, 2020-12-04:
I did update yesterday and still encountered the same issue.

jordanruthe, 2020-12-28:
Are you still having issues with cancelling prints?

jordanruthe, 2021-01-08:
I'm going to close this issue for now since I haven't heard anything. If you are still having issues, please let me know.

closed 2021-01-08 by jordanruthe