KlipperScreen/KlipperScreen #26: incomplete setup on RPI4 (with  KIAUH)

czosnekltd, 2020-12-03:
Unsuccessful installation on Rpi4 2GB RaspbianOS 32bit.
During installation, I get the error -log below:
```
(from ruamel.yaml>=0.11.10->vext>=0.7.4->vext.gi==0.7.4) (0.2.2)
Building wheels for collected packages: vext.gi
  Building wheel for vext.gi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-1ukti3ot
       cwd: /tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/
  Complete output (29 lines):
  running bdist_wheel
  running build
  installing to build/bdist.linux-armv7l/wheel
  running install
  vext.gi Install
  running install_data
  creating build
  creating build/bdist.linux-armv7l
  creating build/bdist.linux-armv7l/wheel
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  copying gi.vext -> build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  running install_egg_info
  running egg_info
  writing vext.gi.egg-info/PKG-INFO
  writing dependency_links to vext.gi.egg-info/dependency_links.txt
  writing requirements to vext.gi.egg-info/requires.txt
  writing top-level names to vext.gi.egg-info/top_level.txt
  reading manifest file 'vext.gi.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.txt'
  writing manifest file 'vext.gi.egg-info/SOURCES.txt'
  Copying vext.gi.egg-info to build/bdist.linux-armv7l/wheel/vext.gi-0.7.4-py3.7.egg-info
  running install_scripts
  Install vext files:
  error: [Errno 2] No such file or directory: 'vext': 'vext'
  ----------------------------------------
  ERROR: Failed building wheel for vext.gi
  Running setup.py clean for vext.gi
Failed to build vext.gi
Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-h59t69wr/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-h59t69wr/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6h7gzwgs/vext-gi_0a516d6aeca54762b34c2627c4a523aa/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-h59t69wr/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.
Installing KlipperScreen unit file
Updating X11 configuration
```

jordanruthe, 2020-12-03:
Have you tried running KlipperScreen despite that error? There is an issue where vext.gi does report an error during installation, but it does install properly and it should still work.

czosnekltd, 2020-12-03:
|      Mainsail: Installed!      |
|                      |     Octoprint: Not installed!  |
|                      |                                |
|                      | KlipperScreen: Incomplete!

jordanruthe, 2020-12-03:
Is there a file /tmp/KlipperScreen.log ? If so, can you attach it

Can you post the output of the following commands?

systemctl status KlipperScreen
journalctl -xe -u KlipperScreen | cat

czosnekltd, 2020-12-03:
pi@raspberrypi:~ $ systemctl status KlipperScreen
● KlipperScreen.service - KlipperScreen
   Loaded: loaded (/etc/systemd/system/KlipperScreen.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
pi@raspberrypi:~ $ journalctl -xe -u KlipperScreen | cat
-- Logs begin at Thu 2020-12-03 17:49:30 GMT, end at Thu 2020-12-03 19:53:01 GMT. --
-- No entries --
pi@raspberrypi:~ $

jordanruthe, 2020-12-03:
Thank you for that output. It looks like KlipperScreen hasn't tried running. I'll push an update to the install script shortly. 

For now, if you run:
sudo systemctl enable KlipperScreen
sudo systemctl start KlipperScreen

It will enable KlipperScreen on startup and start the service. If you have issues with it starting up, please re-run the journalctl command from my last comment.

czosnekltd, 2020-12-03:
thanks
 It works !

closed 2020-12-03 by jordanruthe