KlipperScreen/KlipperScreen #27: Display nozzle temperature in extrusion panel

michelluccote, 2020-12-04:
I was using the extrusion panel and tried to extrude while the nozzle was still warming up. I can see the temp control option on this panel but it would be very useful to see the current temperature in the extrusion panel. In my case I had already set the temp and just had to wait for it to reach this target temp. Having it displayed would save me having to go to the sub panel or another panel to see the current temp.

jordanruthe, 2020-12-04:
The temp should be displayed in the extrude panel, I just pushed a fix to enable it.

michelluccote, 2020-12-04:
Excellent. Your so fast I have to update KlipperScreen between every print! I'll get it updated tomorrow after my 24hr print.

closed 2020-12-05 by jordanruthe