KlipperScreen/KlipperScreen #29: unable to start

minnten, 2020-12-09:
Hello,

if I try to install KlipperScreen via your script, I get an Error and later on, it does not start.

Requirement already satisfied: ruamel.yaml.clib>=0.1.2 in /home/pi/.KlipperScreen-env/lib/python3.7/site-packages (from ruamel.yaml>=0.11.10->vext>=0.7.4->vext.gi==0.7.4) (0.2.2)
Building wheels for collected packages: vext.gi
  Building wheel for vext.gi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/setup.py'"'"'; __file__='"'"'/tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-3vu2uav1
       cwd: /tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/
  Complete output (29 lines):
  running bdist_wheel
  running build
  installing to build/bdist.linux-armv7l/wheel
  running install
  vext.gi Install
  running install_data
  creating build
  creating build/bdist.linux-armv7l
  creating build/bdist.linux-armv7l/wheel
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  copying gi.vext -> build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  running install_egg_info
  running egg_info
  writing vext.gi.egg-info/PKG-INFO
  writing dependency_links to vext.gi.egg-info/dependency_links.txt
  writing requirements to vext.gi.egg-info/requires.txt
  writing top-level names to vext.gi.egg-info/top_level.txt
  reading manifest file 'vext.gi.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.txt'
  writing manifest file 'vext.gi.egg-info/SOURCES.txt'
  Copying vext.gi.egg-info to build/bdist.linux-armv7l/wheel/vext.gi-0.7.4-py3.7.egg-info
  running install_scripts
  Install vext files:
  error: [Errno 2] No such file or directory: 'vext': 'vext'
  ----------------------------------------
  ERROR: Failed building wheel for vext.gi
  Running setup.py clean for vext.gi
Failed to build vext.gi
Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/setup.py'"'"'; __file__='"'"'/tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ij0sl130/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-ij0sl130/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/setup.py'"'"'; __file__='"'"'/tmp/pip-install-o9j192_d/vext-gi_d1926f5bfb3a4624aed5f199a9da5edf/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ij0sl130/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.
Installing KlipperScreen unit file
Created symlink /etc/systemd/system/multi-user.target.wants/KlipperScreen.service → /etc/systemd/system/KlipperScreen.service.
Updating X11 configuration

After the reboot, the screen flashes Black or command promt every 1-2 seconds.

If I try to rerun your script, the error message is gone but the screen flashes again.

I used the newest version of mainsailOS with the latest updates.

Greetings Minnten

jordanruthe, 2020-12-09:
Can you paste the output of the following command?

journalctl -xu KlipperScreen | cat

minnten, 2020-12-09:
`-- Logs begin at Wed 2020-12-09 20:40:49 GMT, end at Wed 2020-12-09 21:17:01 GMT. --
Dec 09 20:46:43 mainsailos systemd[1]: Started KlipperScreen.
-- Subject: A start job for unit KlipperScreen.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit KlipperScreen.service has finished successfully.
-- 
-- The job identifier is 568.
Dec 09 20:46:43 mainsailos xinit[2366]: X.Org X Server 1.20.4
Dec 09 20:46:43 mainsailos xinit[2366]: X Protocol Version 11, Revision 0
Dec 09 20:46:43 mainsailos xinit[2366]: Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Dec 09 20:46:43 mainsailos xinit[2366]: Current Operating System: Linux mainsailos 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l
Dec 09 20:46:43 mainsailos xinit[2366]: Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1024 bcm2708_fb.fbheight=600 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=2a1f4744-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Dec 09 20:46:43 mainsailos xinit[2366]: Build Date: 01 December 2020  05:59:57PM
Dec 09 20:46:43 mainsailos xinit[2366]: xorg-server 2:1.20.4-1+rpt2+deb10u2 (https://www.debian.org/support)
Dec 09 20:46:43 mainsailos xinit[2366]: Current version of pixman: 0.36.0
Dec 09 20:46:43 mainsailos xinit[2366]:         Before reporting problems, check http://wiki.x.org
Dec 09 20:46:43 mainsailos xinit[2366]:         to make sure that you have the latest version.
Dec 09 20:46:43 mainsailos xinit[2366]: Markers: (--) probed, (**) from config file, (==) default setting,
Dec 09 20:46:43 mainsailos xinit[2366]:         (++) from command line, (!!) notice, (II) informational,
Dec 09 20:46:43 mainsailos xinit[2366]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Dec 09 20:46:43 mainsailos xinit[2366]: (==) Log file: "/var/log/Xorg.0.log", Time: Wed Dec  9 20:46:43 2020
Dec 09 20:46:43 mainsailos xinit[2366]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Dec 09 20:46:43 mainsailos xinit[2366]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.4.79-v7+
Dec 09 20:46:47 mainsailos xinit[2366]: INFO:KlipperScreen:Config file: /home/pi/KlipperScreen/KlipperScreen.config
Dec 09 20:46:47 mainsailos xinit[2366]: INFO:KlipperScreen:KlipperScreen version: v0.1.2-38-g9f4204b
Dec 09 20:46:47 mainsailos xinit[2366]: INFO:KlipperScreen.config:Found configuration file at: /home/pi/KlipperScreen/ks_includes/KlipperScreen.conf
Dec 09 20:46:47 mainsailos xinit[2366]: INFO:KlipperScreen.config: ===== Config File =====
Dec 09 20:46:47 mainsailos xinit[2366]: [main]
Dec 09 20:46:47 mainsailos xinit[2366]: [preheat PLA]
Dec 09 20:46:47 mainsailos xinit[2366]: bed = 40
Dec 09 20:46:47 mainsailos xinit[2366]: extruder = 195
Dec 09 20:46:47 mainsailos xinit[2366]: [preheat ABS]
Dec 09 20:46:47 mainsailos xinit[2366]: bed = 90
Dec 09 20:46:47 mainsailos xinit[2366]: extruder = 220
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Main Menu
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main homing]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Homing
Dec 09 20:46:47 mainsailos xinit[2366]: icon = home
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main preheat]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Preheat
Dec 09 20:46:47 mainsailos xinit[2366]: icon = heat-up
Dec 09 20:46:47 mainsailos xinit[2366]: panel = preheat
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main actions]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Actions
Dec 09 20:46:47 mainsailos xinit[2366]: icon = actions
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main config]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Configuration
Dec 09 20:46:47 mainsailos xinit[2366]: icon = control
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main print]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Print
Dec 09 20:46:47 mainsailos xinit[2366]: icon = print
Dec 09 20:46:47 mainsailos xinit[2366]: panel = print
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main homing homeall]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Home All
Dec 09 20:46:47 mainsailos xinit[2366]: icon = home
Dec 09 20:46:47 mainsailos xinit[2366]: method = printer.gcode.script
Dec 09 20:46:47 mainsailos xinit[2366]: params = {"script":"G28"}
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main homing homex]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Home X
Dec 09 20:46:47 mainsailos xinit[2366]: icon = home-x
Dec 09 20:46:47 mainsailos xinit[2366]: method = printer.gcode.script
Dec 09 20:46:47 mainsailos xinit[2366]: params = {"script":"G28 X"}
Dec 09 20:46:47 mainsailos xinit[2366]: [menu __main homing homey]
Dec 09 20:46:47 mainsailos xinit[2366]: name = Home Y
Dec 09 20:46:47 mainsailos xinit[2366]: icon = home-y
Dec 09 20:46:47 mainsailos xinit[2366]: method = printer.gcode.script
Dec 09 20:46:47 mainsailos xinit[2366]: params = {"script":"G28 Y"}
[... 102 lines cut ...]
Dec 09 20:46:47 mainsailos xinit[2366]: INFO:KlipperScreen:Screen resolution: 1024x600
Dec 09 20:46:47 mainsailos xinit[2366]: DEBUG:KlipperScreen.KlippyGtk:img width: 64 height: 64
Dec 09 20:46:47 mainsailos xinit[2366]: DEBUG:KlipperScreen:Loading panel: splash_screen
[... cut ...]
Dec 09 20:46:48 mainsailos xinit[2366]: DEBUG:KlipperScreen.MenuPanel:Evaluating item: homing
Dec 09 20:46:48 mainsailos xinit[2366]: DEBUG:KlipperScreen.MenuPanel:Evaluating item: preheat
Dec 09 20:46:48 mainsailos xinit[2366]: DEBUG:KlipperScreen.MenuPanel:Evaluating item: actions
Dec 09 20:46:48 mainsailos xinit[2366]: DEBUG:KlipperScreen.MenuPanel:Evaluating item: config
Dec 09 20:46:48 mainsailos xinit[2366]: DEBUG:KlipperScreen.MenuPanel:Evaluating item: print
Dec 09 20:46:48 mainsailos xinit[2366]: DEBUG:KlipperScreen:Current panel hierarchy: ['main_panel']`

jordanruthe, 2020-12-09:
It looks like it's starting properly. I see you changed the title... is it working now?

Unfortunately, vext.gi has issues while installing, but it still installs properly for KlipperScreen to be utilized.

minnten, 2020-12-09:
it started now, but I do not know why.
it was my third attempt to install it tomorrow.

the only thing, that was different this time was, that I kept the Pi running for debugging purpose.

minnten, 2020-12-09:
it now works

closed 2020-12-09 by minnten