KlipperScreen/KlipperScreen #34: error requirements.txt

klem1, 2020-12-19:
Hello I copied the scripts to the scripts folder  and with the command 
. /home/pi/klipper/scripts/KlipperScreen-install.sh

I install it everything is ok at the beginning then at the end I have an error, I tried again and always the same.

But a then end i got this message 

```
. /home/pi/klipper/scripts/KlipperScreen-install.sh
dirname: invalid option -- 'b'
Try 'dirname --help' for more information.
Update package data
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
57 packages can be upgraded. Run 'apt list --upgradable' to see them.
Installing package dependencies
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-distutils is already the newest version (3.7.3-1).
python3-gi is already the newest version (3.30.4-1).
python3-gi-cairo is already the newest version (3.30.4-1).
python3-virtualenv is already the newest version (15.1.0+ds-2).
virtualenv is already the newest version (15.1.0+ds-2).
xinit is already the newest version (1.4.0-1).
xinput is already the newest version (1.6.2-1).
gir1.2-gtk-3.0 is already the newest version (3.24.5-1+rpt2).
x11-xserver-utils is already the newest version (7.7+8+rpi1).
xserver-xorg-video-fbturbo is already the newest version (1.20190107~114053).
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
Creating virtual environment
ERROR: Could not open requirements file: [Errno 2] No such file or directory: '/home/pi/scripts/KlipperScreen-requirements.txt'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: vext.gi==0.7.4 in ./.KlipperScreen-env/lib/python3.7/site-packages (0.7.4)
Requirement already satisfied: vext>=0.7.4 in ./.KlipperScreen-env/lib/python3.7/site-packages (from vext.gi==0.7.4) (0.7.4)
Requirement already satisfied: ruamel.yaml>=0.11.10 in ./.KlipperScreen-env/lib/python3.7/site-packages (from vext>=0.7.4->vext.gi==0.7.4) (0.16.12)
Requirement already satisfied: ruamel.yaml.clib>=0.1.2 in ./.KlipperScreen-env/lib/python3.7/site-packages (from ruamel.yaml>=0.11.10->vext>=0.7.4->vext.gi==0.7.4) (0.2.2)
Installing KlipperScreen unit file
-bash: /home/pi/KlipperScreen.service: No such file or directory
The unit files have no installation config (WantedBy=, RequiredBy=, Also=,
Alias= settings in the [Install] section, and DefaultInstance= for template
units). This means they are not meant to be enabled using systemctl.

Possible reasons for having this kind of units are:
• A unit may be statically enabled by being symlinked from another unit's
  .wants/ or .requires/ directory.
• A unit's purpose may be to act as a helper for some other unit which has
  a requirement dependency on it.
• A unit may be started when needed via activation (socket, path, timer,
  D-Bus, udev, scripted systemctl call, ...).
• In case of template units, the unit is meant to be enabled with some
  instance name specified.
Updating X11 configuration
Failed to start KlipperScreen.service: Unit KlipperScreen.service has a bad unit file setting.
See system logs and 'systemctl status KlipperScreen.service' for details.

```

jordanruthe, 2020-12-19:
You shouldn't copy the file anywhere. Once you have cloned the repository, run the script from the KlipperScreen scripts folder.

klem1, 2020-12-19:
i clone your folder here  . /home/pi/KlipperScreen-master/

and start the scripts

 . /home/pi/KlipperScreen-master/scripts/KlipperScreen-install.sh

```
. /home/pi/KlipperScreen-master/scripts/KlipperScreen-install.                                                           sh
dirname: invalid option -- 'b'
Try 'dirname --help' for more information.
Update package data
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
57 packages can be upgraded. Run 'apt list --upgradable' to see them.
Installing package dependencies
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-distutils is already the newest version (3.7.3-1).
python3-gi is already the newest version (3.30.4-1).
python3-gi-cairo is already the newest version (3.30.4-1).
python3-virtualenv is already the newest version (15.1.0+ds-2).
virtualenv is already the newest version (15.1.0+ds-2).
xinit is already the newest version (1.4.0-1).
xinput is already the newest version (1.6.2-1).
gir1.2-gtk-3.0 is already the newest version (3.24.5-1+rpt2).
x11-xserver-utils is already the newest version (7.7+8+rpi1).
xserver-xorg-video-fbturbo is already the newest version (1.20190107~114053).
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
Creating virtual environment
ERROR: Could not open requirements file: [Errno 2] No such file or directory: '/home/pi/scripts/KlipperScr                                 een-requirements.txt'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: vext.gi==0.7.4 in ./.KlipperScreen-env/lib/python3.7/site-packages (0.7.4)
Requirement already satisfied: vext>=0.7.4 in ./.KlipperScreen-env/lib/python3.7/site-packages (from vext.                                 gi==0.7.4) (0.7.4)
Requirement already satisfied: ruamel.yaml>=0.11.10 in ./.KlipperScreen-env/lib/python3.7/site-packages (f                                 rom vext>=0.7.4->vext.gi==0.7.4) (0.16.12)
Requirement already satisfied: ruamel.yaml.clib>=0.1.2 in ./.KlipperScreen-env/lib/python3.7/site-packages                                  (from ruamel.yaml>=0.11.10->vext>=0.7.4->vext.gi==0.7.4) (0.2.2)
Installing KlipperScreen unit file
-bash: /home/pi/KlipperScreen.service: No such file or directory
The unit files have no installation config (WantedBy=, RequiredBy=, Also=,
Alias= settings in the [Install] section, and DefaultInstance= for template
units). This means they are not meant to be enabled using systemctl.

Possible reasons for having this kind of units are:
• A unit may be statically enabled by being symlinked from another unit's
  .wants/ or .requires/ directory.
• A unit's purpose may be to act as a helper for some other unit which has
  a requirement dependency on it.
• A unit may be started when needed via activation (socket, path, timer,
  D-Bus, udev, scripted systemctl call, ...).
• In case of template units, the unit is meant to be enabled with some
  instance name specified.
Updating X11 configuration
Failed to start KlipperScreen.service: Unit KlipperScreen.service has a bad unit file setting.
See system logs and 'systemctl status KlipperScreen.service' for details.

```

jordanruthe, 2020-12-19:
Run it from the scripts directory:
cd /home/pi/KlipperScreen-master/scripts
./KlipperScreen-install.sh

klem1, 2020-12-19:
Oh ok  but i got this 

`-bash: ./KlipperScreen-install.sh: Permission denied`

what is the command to have the permission.

jordanruthe, 2020-12-19:
chmod +x KlipperScreen-install.sh

klem1, 2020-12-19:
oh god it's working

i got  Klipper screen initializing

klem1, 2020-12-19:
thanks i will try to find something to install my hdmi touch screen.

closed 2020-12-20 by jordanruthe