KlipperScreen/KlipperScreen #36: Status screen not updating

mninja79, 2020-12-21:
Hi,
Print job status screen is not updating. I get this error:
` self.panels[sub].process_update(action, data)
Dec 21 11:43:24 Printer xinit[30095]:   File "/home/pi/KlipperScreen/panels/job_status.py", line 351, in process_update
Dec 21 11:43:24 Printer xinit[30095]:     self._printer.get_dev_stat(x,"target")
Dec 21 11:43:24 Printer xinit[30095]:   File "/home/pi/KlipperScreen/panels/job_status.py", line 506, in update_temp
Dec 21 11:43:24 Printer xinit[30095]:     self.labels[x].set_markup(
Dec 21 11:43:24 Printer xinit[30095]: KeyError: 'extruder1'
Dec 21 11:43:24 Printer xinit[30095]: Traceback (most recent call last):
Dec 21 11:43:24 Printer xinit[30095]:   File "/home/pi/KlipperScreen/screen.py", line 401, in _websocket_callback
Dec 21 11:43:24 Printer xinit[30095]:     self.panels[sub].process_update(action, data)
Dec 21 11:43:24 Printer xinit[30095]:   File "/home/pi/KlipperScreen/panels/job_status.py", line 351, in process_update
Dec 21 11:43:24 Printer xinit[30095]:     self._printer.get_dev_stat(x,"target")
Dec 21 11:43:24 Printer xinit[30095]:   File "/home/pi/KlipperScreen/panels/job_status.py", line 506, in update_temp
Dec 21 11:43:24 Printer xinit[30095]:     self.labels[x].set_markup(
Dec 21 11:43:24 Printer xinit[30095]: KeyError: 'extruder1'`

It is extruder that is selected but it is checking extruder1
I have two extruders sharing one hotend.  When the printer is on idle it always shows tool2 temp even when tool1 is selected.

Thanks

jordanruthe, 2020-12-21:
I'm not sure what you meant by "When the printer is on idle it always shows tool2 temp even when tool1 is selected"

I just pushed some updates. The idle page should now have the temp shown for multiple extruders. The job status page should now show the currently active extruder.

jordanruthe, 2020-12-21:
I also updated the extrude panel under actions to show multiple extruders and allow switching between them. It does use the T0/T1 style commands, however. Let me know if you have any issues with that panel.

mninja79, 2020-12-21:
> I'm not sure what you meant by "When the printer is on idle it always shows tool2 temp even when tool1 is selected"
> 
> I just pushed some updates. The idle page should now have the temp shown for multiple extruders. The job status page should now show the currently active extruder.

mninja79, 2020-12-21:
pulled the latest commit and the job status panel is updating. Thanks
I'll update you  on the main menu panel once the print is complete.

mninja79, 2020-12-22:
Main menu panel works fine now thanks. Only issue is job status does not come back up after one print is complete or if I cancel a print.

jordanruthe, 2020-12-22:
Do you have the log file from when the status panel doesn't pop up?

mninja79, 2020-12-22:
Here a the klipper screen logs.
[KlipperScreen1.log](https://github.com/jordanruthe/KlipperScreen/files/5731213/KlipperScreen1.log)

jordanruthe, 2020-12-22:
Thank you for the log file. I'm having a hard time telling exactly what's happening.
When it's not showing, what is displayed on the screen?
Are you attempting to print the same or different gcode file?
I'm trying to figure out how I can recreate this situation as I'm able to print multiple files and have not ran into that.

mninja79, 2020-12-22:
It stays on the main menu panel. Or what ever panel I was last on e.g I had it on temperature menu and it did not change when I started printing. I have to restart the KlipperScreen. One weird thing gcode files ever show up in the  print panel.  
What I'll do is reinstall the KlipperScreen and get back to you.

mninja79, 2020-12-22:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5731958/KlipperScreen.log)
I reinstalled it now I have to restart program for the job status screen to load.

jordanruthe, 2020-12-28:
I made some changes today. Can you see if this helps?

mninja79, 2020-12-29:
Still have the same issue. I have been getting this error when I restart KlipperScreen.
`Dec 29 16:10:56 Printer xinit[17151]:     f = open("%s/%s-%s" % (self.thumbnail_dir, params['filename'], thumbnail['size']), "wb")
Dec 29 16:10:56 Printer xinit[17151]: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/.KS-thumbnails/PLA/nozzle.6/tram/rooftop_center_main_0.32mm_PLA__47m.gcode-34016'
Dec 29 16:10:56 Printer xinit[17151]: DEBUG:KlipperScreen.KlippyFiles:Got metadata for PLA/nozzle.6/tram/rooftop_end_main_0.32mm_PLA__59m.gcode
Dec 29 16:10:56 Printer xinit[17151]: Traceback (most recent call last):
Dec 29 16:10:56 Printer xinit[17151]:   File "/home/pi/KlipperScreen/ks_includes/files.py", line 75, in _callback
Dec 29 16:10:56 Printer xinit[17151]:     f = open("%s/%s-%s" % (self.thumbnail_dir, params['filename'], thumbnail['size']), "wb")
Dec 29 16:10:56 Printer xinit[17151]: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/.KS-thumbnails/PLA/nozzle.6/tram/rooftop_end_main_0.32mm_PLA__59m.gcode-49168'
Dec 29 16:10:56 Printer xinit[17151]: DEBUG:KlipperScreen.KlippyFiles:Got metadata for PLA/nozzle.6/tram/seat_frame_0.2mm_PLA__24m.gcode
Dec 29 16:10:56 Printer xinit[17151]: Traceback (most recent call last):
Dec 29 16:10:56 Printer xinit[17151]:   File "/home/pi/KlipperScreen/ks_includes/files.py", line 75, in _callback
Dec 29 16:10:56 Printer xinit[17151]:     f = open("%s/%s-%s" % (self.thumbnail_dir, params['filename'], thumbnail['size']), "wb")
Dec 29 16:10:56 Printer xinit[17151]: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/.KS-thumbnails/PLA/nozzle.6/tram/seat_frame_0.2mm_PLA__24m.gcode-45644'
Dec 29 16:10:56 Printer xinit[17151]: DEBUG:KlipperScreen.KlippyFiles:Got metadata for PLA/nozzle.6/tram/seat_insert_0.2mm_PLA__13m.gcode
Dec 29 16:10:56 Printer xinit[17151]: Traceback (most recent call last):
Dec 29 16:10:56 Printer xinit[17151]:   File "/home/pi/KlipperScreen/ks_includes/files.py", line 75, in _callback
Dec 29 16:10:56 Printer xinit[17151]:     f = open("%s/%s-%s" % (self.thumbnail_dir, params['filename'], thumbnail['size']), "wb")
Dec 29 16:10:56 Printer xinit[17151]: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/.KS-thumbnails/PLA/nozzle.6/tram/seat_insert_0.2mm_PLA__13m.gcode-35668'
Dec 29 16:10:56 Printer xinit[17151]: DEBUG:KlipperScreen.KlippyFiles:Got metadata for PLA/nozzle.6/tram/Wheel-Oring_0.15mm_PLA__24m.gcode
Dec 29 16:10:56 Printer xinit[17151]: Traceback (most recent call last):
Dec 29 16:10:56 Printer xinit[17151]:   File "/home/pi/KlipperScreen/ks_includes/files.py", line 75, in _callback
Dec 29 16:10:56 Printer xinit[17151]:     f = open("%s/%s-%s" % (self.thumbnail_dir, params['filename'], thumbnail['size']), "wb")
Dec 29 16:10:56 Printer xinit[17151]: FileNotFoundError: [Errno 2] No such file or directory: '/tmp/.KS-thumbnails/PLA/nozzle.6/tram/Wheel-Oring_0.15mm_PLA__24m.gcode-40964'
Dec 29 16:10:56 Printer xinit[17151]: DEBUG:KlipperScreen.KlippyFiles:Got metadata for PLA/nozzle.6/tram/Wheel_0.2mm_PLA__18m.gcode`
The errors above are from /var/log/daemon.log
There are no thumbnails in  /tmp/.KS-thumbnails directory.

When tail the Klipperscreen.log and restart I see it is stops logging after a couple of  get metadata from files in /sdcard. 2020-12-29 16:10:56,897 - KlipperScreen.KlippyFiles - DEBUG - Got metadata for PLA/nozzle.6/tram/Wheel_0.2mm_PLA__18m.gcode

[image]

There are no files in my print panel. 
Are there any specific logs you require?

jordanruthe, 2020-12-29:
Ah, thanks for that. I haven't built in directory support yet, unfortunately. I just pushed an update that should make the thumbnails work for you and get rid of the file error from above.

mninja79, 2020-12-29:
Great work! All seems to be working. I'll update the job status issue in a few when I'm printing. Thanks for the quick response.

mninja79, 2020-12-29:
job status screen working. cheers mate!

closed 2020-12-29 by mninja79