KlipperScreen/KlipperScreen #37: No Job Status Screen

robvoi, 2020-12-27:
Hi,

first of - thanks a lot for your work and for sharing the results!
I use your solution for the printer I currently build.

I have an issue though. When I start a print the display doesn't switch to the job status.

Possible causes - in case this is not an issue with the software:
- I use a 800*600 display (works fine)
- I have not yet connected any heaters nor do I extrude filament. I just use X/Y axis test code. (move to 0,0 -> move to 300,300 -> repeat)
- I use mainsail in parallel

Any idea on what might be wrong?

robvoi, 2020-12-28:
Here a fresh set of logs & config (copy from logs in #38)
[printer.cfg.txt](https://github.com/jordanruthe/KlipperScreen/files/5746651/printer.cfg.txt)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5746652/klippy.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5746653/KlipperScreen.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5746654/moonraker.log)

robvoi, 2020-12-28:
I am still building my printer. So no real print job runs so far. With my test job (attached) KlipperScreen doesn't switch to the job status panel when printing.

[test.gcode.txt](https://github.com/jordanruthe/KlipperScreen/files/5748971/test.gcode.txt)

jordanruthe, 2020-12-28:
Can you post another KlipperScreen.log where you try to print with the latest version?

robvoi, 2020-12-29:
Hi. Here a log after the print with the newest version. No job status panel showed up during the print.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5751986/KlipperScreen.log)

robvoi, 2021-01-02:
With gcode from a "normal" print the panel shows up. Not sure what is missing in the test gcode to trigger the panel. You may close the issue as it doesn't seem to be a problem in real life.

closed 2021-01-03 by jordanruthe

robvoi, 2021-01-11:
I have another print file where KlipperScreen doesn't switch to the print job panel.
Very low priority. But maybe you can have a look.
Tried to find what triggers the panel/printing status but didn't find it in the code.
[pressure_advance2.gcode.txt](https://github.com/jordanruthe/KlipperScreen/files/5798295/pressure_advance2.gcode.txt)