KlipperScreen/KlipperScreen #38: No Tool / nozzle temperature

robvoi, 2020-12-27:
KlipperScreen shows no tool / tool temparature.
Log states: Toolcount: 0

I installed using the KIAUH script. Current KlipperScreen verison is 0.1.2-58

Any idea?

jordanruthe, 2020-12-27:
Can you post your entire klipper config or the klipper config log?

robvoi, 2020-12-28:
Thanks for looking into it!
Here fresh logs and the printer.cfg. All after a fresh reboot with no further actions.

[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5746640/moonraker.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5746641/klippy.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5746643/KlipperScreen.log)
[printer.cfg.txt](https://github.com/jordanruthe/KlipperScreen/files/5746644/printer.cfg.txt)

robvoi, 2020-12-28:
Small update: the main screen doesn't show the extruder. In the extrude panel I see the extruder. Also with its temperature.
There is just a very big delay (in minutes) on showing the actual values.

jordanruthe, 2020-12-28:
Can you update and re-run? I've made some changes but also included some debugs which would help me narrow down this problem.

robvoi, 2020-12-28:
Great that you are on it. KlipperScreen would be a great feature for my printer.
Updated and tested. No change with the new version (also not for the other issues, which may all be connected).
Fresh logs attached.

[KlipperScreen_after_klipper_restart_(now_pass_init).log](https://github.com/jordanruthe/KlipperScreen/files/5748811/KlipperScreen_after_klipper_restart_.now_pass_init.log)
[KlipperScreen_after_reboot.log](https://github.com/jordanruthe/KlipperScreen/files/5748812/KlipperScreen_after_reboot.log)

jordanruthe, 2020-12-28:
Do you have a heated bed on your printer?

robvoi, 2020-12-28:
Not yet installed. Can be done tomorrow though.

jordanruthe, 2020-12-28:
I wasn't accounting for no heated bed. I just pushed another change that fixes several panels for your case.

robvoi, 2020-12-28:
Great. This solved the issue. Same for the init problem.
So far only only the job status part #37 is left. I will update this one and close the others.

Again - thanks for the effort you put into this project.

closed 2020-12-28 by robvoi