KlipperScreen/KlipperScreen #39: initilizing on boot

robvoi, 2020-12-27:
When the system boots KlipperScreen stays on "in initializing".
When I press "Klipper restart" klipper restarts and the screen switches to main menu.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5745206/KlipperScreen.log)

robvoi, 2020-12-28:
Here a KlipperScreen log after fresh reboot with the screen stuck at "initilizing".
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5746635/KlipperScreen.log)

Tomblarom, 2020-12-28:
I'm experiencing exactly the same issue. Installed all recent versions and KlipperScreen won't switch to the panels. Octoprint is working fine. Restarted all services one by one..

Stuck on `Initializing`. Selecting `Firmware Restart` and it's stuck on `Klipper has shutdown`.

Klipper v0.9.1-139 [[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5746951/klippy.log)]
Moonraker v0.2.6-3 [[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5746952/moonraker.log)]
KlipperScreen v0.1.2-58 [[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5746953/KlipperScreen.log)]

robvoi, 2020-12-28:
Solved with new version based in issue #38 (no heated bed).

closed 2020-12-28 by robvoi