KlipperScreen/KlipperScreen #42: No Move Actions after updates

robvoi, 2021-01-09:
I updated Klipper, Moonraker and Klipperscreen.
Now the move actions (X+,X-,Y+ ...) don't do anything anymore. The rest seems to be fine. (homing, heating etc.)
Control via Klipperscreen works. 

Klipper 0.9.1-143
Moonraker 0.2.8-27
Mainsail 0.4.1
KlipperScreen 0.1.2-85

[moonraker_noMove.log](https://github.com/jordanruthe/KlipperScreen/files/5791380/moonraker_noMove.log)
[klippy_noMove.log](https://github.com/jordanruthe/KlipperScreen/files/5791381/klippy_noMove.log)
[KlipperScreen_noMove.log](https://github.com/jordanruthe/KlipperScreen/files/5791382/KlipperScreen_noMove.log)

jordanruthe, 2021-01-09:
Good catch, sorry about that. Just pushed an update.

robvoi, 2021-01-09:
Wow. Lightspeed support. Thanks a lot. This solved the problem.

closed 2021-01-09 by robvoi