KlipperScreen/KlipperScreen #44: Installation Fails

nicedevil007, 2021-01-10:
Heya, 

just run your script after a ```git clone``` to install your KlipperScreen.
It returns the following. Is there a fix?

```Building wheels for collected packages: vext.gi
  Building wheel for vext.gi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /root/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-pitomfw3
       cwd: /tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/
  Complete output (29 lines):
  running bdist_wheel
  running build
  installing to build/bdist.linux-armv7l/wheel
  running install
  vext.gi Install
  running install_data
  creating build
  creating build/bdist.linux-armv7l
  creating build/bdist.linux-armv7l/wheel
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  copying gi.vext -> build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  running install_egg_info
  running egg_info
  writing vext.gi.egg-info/PKG-INFO
  writing dependency_links to vext.gi.egg-info/dependency_links.txt
  writing requirements to vext.gi.egg-info/requires.txt
  writing top-level names to vext.gi.egg-info/top_level.txt
  reading manifest file 'vext.gi.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.txt'
  writing manifest file 'vext.gi.egg-info/SOURCES.txt'
  Copying vext.gi.egg-info to build/bdist.linux-armv7l/wheel/vext.gi-0.7.4-py3.7.egg-info
  running install_scripts
  Install vext files:
  error: [Errno 2] No such file or directory: 'vext': 'vext'
  ----------------------------------------
  ERROR: Failed building wheel for vext.gi
  Running setup.py clean for vext.gi
Failed to build vext.gi
Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /root/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-587refqp/install-record.txt --single-version-externally-managed --compile --install-headers /root/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /root/.KlipperScreen-env/share/vext
    creating /root/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /root/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /root/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-587refqp/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /root/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8tgs5uvr/vext-gi_7b15bbdfbf054ca4a53aea893eb54224/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-587refqp/install-record.txt --single-version-externally-managed --compile --install-headers /root/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.
Installing KlipperScreen unit file
```

jordanruthe, 2021-01-10:
I have been unable to find a way to suppress vext.gi installation "errors." Despite this, it should still work properly.

nicedevil007, 2021-01-10:
after a reboot, it seems to be working, yeah :) sry for opening an issue... on first try it seems like it wouldn't :(

Btw: Just if you want to add it to your readme/description.

Raspberry Pi screen works well with the lower resolution :)

I'm using these settings in the `config.txt`

```
framebuffer_width=1024
framebuffer_height=614

hdmi_cvt=800 480 60 6 0 0 1
hdmi_group=2
hdmi_mode=87
hdmi_drive=2
```

closed 2021-01-10 by nicedevil007