KlipperScreen/KlipperScreen #45: Feature Request

nicedevil007, 2021-01-10:
Heya again, now, after your plugin/addon seems to work as intended I found a small thing that is a bit annoying after you uploaded a lot of files to your pi :)

The printable lilelist is sorted by name but I would say, most of us want to see the most recent files at the top of the list instead of scrolling through the whole list or?

Maybe you could add a sort option by name or upload date?

jordanruthe, 2021-01-13:
This feature will be added in the near future

jordanruthe, 2021-01-13:
This should be resolved by a commit now.

hunagyp, 2021-01-13:
@jordanruthe
https://www.youtube.com/watch?v=oKpGP7Snz70
At some point, the list ordering does not change despite the fact, I click on the order buttons back-and-forth..
Can you take a look on the cause?
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5811416/KlipperScreen.log)

jordanruthe, 2021-01-13:
From your video, it looks like it's working properly. This is a weird case where the order of the names is the same order of the date, so it looks like nothing is happening.

hunagyp, 2021-01-14:
You're right, shame on me I did not realize this... Sorry for wasting your time!

closed 2021-01-22 by jordanruthe