KlipperScreen/KlipperScreen #46: Feature Request Z_Tilt Adjust Button in Bed Level....

EraserFX, 2021-01-10:
Feature Request: A Button for Z_Tilt Adjust in bed level section would be great. And Gantry Leveling too.
GCode Macros Visible and invisible Option.

jordanruthe, 2021-01-10:
I'm not sure how z tilt or gantry leveling works, as my setup does not have these features. If you can provide info on how these work/what gcode commands to run I can see about adding it.

Gcode macros are hideable from the settings page.

EraserFX, 2021-01-10:
https://www.klipper3d.org/Config_Reference.html#bed_tilt

`[z_tilt]
Multiple Z stepper tilt adjustment. This feature enables independent adjustment of multiple z steppers (see the “stepper_z1” section) to adjust for tilt. If this section is present then a Z_TILT_ADJUST extended G-Code command becomes available.

[z_tilt]
#z_positions:
#   A list of X,Y coordinates (one per line; subsequent lines
#   indented) describing the location of each bed "pivot point". The
#   "pivot point" is the point where the bed attaches to the given Z
#   stepper. It is described using nozzle coordinates (the XY position
#   of the nozzle if it could move directly above the point). The
#   first entry corresponds to stepper_z, the second to stepper_z1,
#   the third to stepper_z2, etc. This parameter must be provided.
#points:
#   A list of X,Y coordinates (one per line; subsequent lines
#   indented) that should be probed during a Z_TILT_ADJUST command.
#   Specify coordinates of the nozzle and be sure the probe is above
#   the bed at the given nozzle coordinates. This parameter must be
#   provided.
#speed: 50
#   The speed (in mm/s) of non-probing moves during the calibration.
#   The default is 50.
#horizontal_move_z: 5
#   The height (in mm) that the head should be commanded to move to
#   just prior to starting a probe operation. The default is 5.
#retries: 0
#   Number of times to retry if the probed points aren't within
#   tolerance.
#retry_tolerance: 0
#   If retries are enabled then retry if largest and smallest probed
#   points differ more than retry_tolerance. Note the smallest unit of
#   change here would be a single step. However if you are probing
#   more points than steppers then you will likely have a fixed
#   minimum value for the range of probed points which you can learn
#   by observing command output.`

`[quad_gantry_level]
Moving gantry leveling using 4 independently controlled Z motors. Corrects hyperbolic parabola effects (potato chip) on moving gantry which is more flexible. WARNING: Using this on a moving bed may lead to undesirable results. If this section is present then a QUAD_GANTRY_LEVEL extended G-Code command becomes available. This routine assumes the following Z motor configuration:

 ----------------
 |Z1          Z2|
 |  ---------   |
 |  |       |   |
 |  |       |   |
 |  x--------   |
 |Z           Z3|
 ----------------
Where x is the (0,0) point on the bed

[quad_gantry_level]
#gantry_corners:
#   A newline separated list of X,Y coordinates describing the two
#   opposing corners of the gantry. The first entry corresponds to Z,
#   the second to Z2. This parameter must be provided.
#points:
#   A newline separated list of four X,Y points that should be probed
#   during a QUAD_GANTRY_LEVEL command. Order of the locations is
#   important, and should correspond to Z, Z1, Z2, and Z3 location in
#   order. This parameter must be provided. For maximum accuracy,
#   ensure your probe offsets are configured.
#speed: 50
#   The speed (in mm/s) of non-probing moves during the calibration.
#   The default is 50.
#horizontal_move_z: 5
#   The height (in mm) that the head should be commanded to move to
#   just prior to starting a probe operation. The default is 5.
#max_adjust: 4
#   Safety limit if an adjustment greater than this value is requested
#   quad_gantry_level will abort.
#retries: 0
#   Number of times to retry if the probed points aren't within
#   tolerance.
#retry_tolerance: 0
#   If retries are enabled then retry if largest and smallest probed
#   points differ more than retry_tolerance.`

EraserFX, 2021-01-10:
https://www.klipper3d.org/Config_Reference.html#bed_tilt

EraserFX, 2021-01-10:
[z_tilt]
Multiple Z stepper tilt adjustment. This feature enables independent adjustment of multiple z steppers (see the “stepper_z1” section) to adjust for tilt. If this section is present then a Z_TILT_ADJUST extended G-Code command becomes available.
Voron 1.8 for example

[quad_gantry_level]
Moving gantry leveling using 4 independently controlled Z motors. Corrects hyperbolic parabola effects (potato chip) on moving gantry which is more flexible. WARNING: Using this on a moving bed may lead to undesirable results. If this section is present then a QUAD_GANTRY_LEVEL extended G-Code command becomes available. This routine assumes the following Z motor configuration. <-- Voron V2 for Example

jordanruthe, 2021-01-31:
quad_gantry_level should now appear under the actions menu. Keep this open for the z_tilt function, however

nicedevil007, 2021-02-21:
for `z_tilt` functionality you just have to add the gcode macro

`Z_TILT_ADJUST`

to the button and make sure that the printer is homed first.

This is done completly automatical then if setup correctly in the `printer.cfg`. So just a button do start this gcode would be enough in my opinion? Don't know if you are able to parse the console output of the procedure so we can notice what the difference between right and left is at this point.

jordanruthe, 2021-02-21:
What is the output of the z_tilt_adjust command? It'd be helpful if you could paste the console output from where you enter a command to the end of the output.

Raabi91, 2021-02-22:
> What is the output of the z_tilt_adjust command? It'd be helpful if you could paste the console output from where you enter a command to the end of the output.

it looks like it but is actually completely automatic

18:41:27
$ Z_TILT_ADJUST
18:41:29
// probe at 90.000,175.000 is z=2.695000
18:41:30
// probe at 90.000,175.000 is z=2.687500
18:41:34
// probe at 320.000,175.000 is z=2.267500
18:41:36
// probe at 320.000,175.000 is z=2.270000
18:41:36
// Making the following Z adjustments:
// stepper_z = -0.511573
// stepper_z1 = 0.425211
18:41:36
// Retries: 0/10 Probed points range: 0.422500 tolerance: 0.025000
18:41:39
// probe at 90.000,175.000 is z=2.484073
18:41:41
// probe at 90.000,175.000 is z=2.499073
18:41:44
// probe at 320.000,175.000 is z=2.521573
18:41:46
// probe at 320.000,175.000 is z=2.484073
18:41:46
// Making the following Z adjustments:
// stepper_z = -0.022794
// stepper_z1 = -0.047728
18:41:46
// Making the following Z adjustments:
// stepper_z = -0.022794
// stepper_z1 = -0.047728
18:41:46
// Retries: 1/10 Probed points range: 0.011250 tolerance: 0.025000
18:41:46
// Retries: 1/10 Probed points range: 0.011250 tolerance: 0.025000

jordanruthe, 2021-02-22:
I'm trying to think about where to place these. I'm thinking QUAD_GANTRY_LEVEL and Z_TILT_ADJUST belong in Homing... any arguments for or against this?

As for the output, I'm going to say I'm not going to parse any of the output from the command. I plan to introduce a console view in the future, and if needed that could be used to look at the output.

Raabi91, 2021-02-22:
sounds good i currently have my z tilt placed in homeing. but what would be more interesting is to check if it is in the clipper config and only display it if it is.

Elektrickser, 2021-02-22:
@jordanruthe 

Yeah, that sounds good. I was just about to implement that.

closed 2021-02-22 by jordanruthe

Elektrickser, 2021-02-24:
Hi @jordanruthe ,
i don't become the menu-Item for Z-Tilt, I will send my Klipper-log and config. 
And i make a new svg-Logo for z-Tilt. I will uploud it too.
[z-tilt.zip](https://github.com/jordanruthe/KlipperScreen/files/6039365/z-tilt.zip)

[KlipperScreen(1).zip](https://github.com/jordanruthe/KlipperScreen/files/6039368/KlipperScreen.1.zip)

commit 57b31806 (2021-02-22): printer: Add quad_gantry_level/z_tilt_adjust to homing menu. Fixes #46
commit f6f2f3f1 (2021-02-22): printer: Add quad_gantry_level/z_tilt_adjust to homing menu. Fixes #46