KlipperScreen/KlipperScreen #49: macros visibility does not work as expected

hunagyp, 2021-01-11:
There are items which are selected to be hidden but are still shown.
END_PRINT, M600,PURGE_LINE_BEFORE_PRINT
And some are set to be shown but are not visible:
CANCEL_PRINT,LOAD_FILAMENT,PURGE,PARK...

My Macros screen:
[image]

Macros visibility settings:
[https://www.youtube.com/watch?v=cWiBNtBtAp4](url)

[KlipperScreen.conf.txt](https://github.com/jordanruthe/KlipperScreen/files/5798517/KlipperScreen.conf.txt)
[klippy (2).log.txt](https://github.com/jordanruthe/KlipperScreen/files/5798518/klippy.2.log.txt)
[KlipperScreen.log.txt](https://github.com/jordanruthe/KlipperScreen/files/5798519/KlipperScreen.log.txt)

jordanruthe, 2021-01-13:
I've made some changes to the macro panel which should fix this problem.

hunagyp, 2021-01-13:
@jordanruthe Thank you a lot, I tried this patch and the macros are correctly shown/hidden now 👍

closed 2021-01-14 by jordanruthe