KlipperScreen/KlipperScreen #50: [Feature Request] Allow assignment of screws to position in `screws_tild_calculate`

tbe, 2021-01-12:
In my setup, the fixed screw is back left, but the display shows it as back right.

As this setup may vary for users, it should be possible to map the screw names to their display position.

jordanruthe, 2021-01-13:
This is a bug, the fixed position should be automatically detected. Can you post a /tmp/KlipperScreen.log file?

tbe, 2021-01-13:
It look like that the order of  `screws_tilt_adjust` and `bed_screws` has to match ....

```
2021-01-12 17:52:07,103 - KlipperScreen - DEBUG - Current panel hierarchy: ['main_panel', 'main_panel_config']
2021-01-12 17:52:08,544 - KlipperScreen - DEBUG - Loading panel: bed_level
2021-01-12 17:52:08,544 - KlipperScreen - INFO - Panel path: /home/printer/KlipperScreen/panels/bed_level.py
2021-01-12 17:52:08,558 - KlipperScreen.ScreenPanel - DEBUG - Button scale: [0.625, 0.625]
2021-01-12 17:52:08,572 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: back_left
2021-01-12 17:52:08,573 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: front_left
2021-01-12 17:52:08,573 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: back_right
2021-01-12 17:52:08,574 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: 244, 216
2021-01-12 17:52:08,574 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: 244, 44
2021-01-12 17:52:08,574 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: 76, 216
2021-01-12 17:52:08,574 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: front_right
2021-01-12 17:52:08,574 - KlipperScreen.BedLevelPanel - DEBUG - Screws section: 76, 44
2021-01-12 17:52:08,575 - KlipperScreen.BedLevelPanel - DEBUG - Bed screw locations [x,y]: [[76.0, 44.0], [244.0, 44.0], [76.0, 216.0], [244.0, 216.0]]
2021-01-12 17:52:08,575 - KlipperScreen.BedLevelPanel - DEBUG - Screws: [[31.7, 51.0], [199.7, 51.0], [31.7, 223.0], [199.7, 223.0]]
2021-01-12 17:52:08,575 - KlipperScreen.BedLevelPanel - DEBUG - Configured screw locations [x,y]: [[31.7, 51.0], [199.7, 51.0], [31.7, 223.0], [199.7, 223.0]]
2021-01-12 17:52:08,606 - KlipperScreen - DEBUG - Current panel hierarchy: ['main_panel', 'main_panel_config', 'main_panel_config_bed_level']
2021-01-12 17:52:10,971 - KlipperScreen.KlipperWebsocket - DEBUG - Sending printer.gcode.script: SCREWS_TILT_CALCULATE
...
2021-01-12 17:52:38,741 - KlipperScreen.BedLevelPanel - DEBUG - // back_right : X 244.0, Y 216.0, Z 2.16500 : Adjust -> CW 00:36
2021-01-12 17:52:38,741 - KlipperScreen.BedLevelPanel - DEBUG - X: 199 Y: 223
2021-01-12 17:52:38,742 - KlipperScreen.BedLevelPanel - DEBUG - [31.7, 51.0]
2021-01-12 17:52:38,742 - KlipperScreen.BedLevelPanel - DEBUG - [199.7, 51.0]
2021-01-12 17:52:38,742 - KlipperScreen.BedLevelPanel - DEBUG - [31.7, 223.0]
2021-01-12 17:52:38,743 - KlipperScreen.BedLevelPanel - DEBUG - [199.7, 223.0]
2021-01-12 17:52:38,745 - KlipperScreen - DEBUG - [
  "notify_gcode_response",
  "// front_right : X 244.0, Y 44.0, Z 2.10500 : Adjust -> CW 00:43"
]
2021-01-12 17:52:38,745 - KlipperScreen.BedLevelPanel - DEBUG - // front_right : X 244.0, Y 44.0, Z 2.10500 : Adjust -> CW 00:43
2021-01-12 17:52:38,745 - KlipperScreen.BedLevelPanel - DEBUG - X: 199 Y: 51
2021-01-12 17:52:38,745 - KlipperScreen.BedLevelPanel - DEBUG - [31.7, 51.0]
2021-01-12 17:52:38,745 - KlipperScreen.BedLevelPanel - DEBUG - [199.7, 51.0]
2021-01-12 17:52:38,747 - KlipperScreen - DEBUG - [
  "notify_gcode_response",
  "// front_left : X 76.0, Y 44.0, Z 2.52000 : Adjust -> CCW 00:07"
]
2021-01-12 17:52:38,747 - KlipperScreen.BedLevelPanel - DEBUG - // front_left : X 76.0, Y 44.0, Z 2.52000 : Adjust -> CCW 00:07
2021-01-12 17:52:38,747 - KlipperScreen.BedLevelPanel - DEBUG - X: 31 Y: 51
2021-01-12 17:52:38,747 - KlipperScreen.BedLevelPanel - DEBUG - [31.7, 51.0]
```

I reordered the `bed_screws` block, and will test it after the current print.

tbe, 2021-01-15:
Nope, didn't do the trick. Here is a clean Logfile from just one adjust attempt.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5819007/KlipperScreen.log)

commit f1ced2ac (2021-01-26): bed_level: Fix #50
closed 2021-01-26 by jordanruthe