KlipperScreen/KlipperScreen #53: [FR] time left 0 while still printing

hunagyp, 2021-01-14:
Is it possible to show that kind of progress bar, what can be seen on a character based LCD?
I know, (sum)print time estimation is a hard thing, that's why(?) KS shows remaining time 0 min, while the print job is still running. Maybe an additional progress bar would add a "touch" the UIX :)
Maybe even "Time left" label could  count and show the additional time, after it reaches 0
For example
[hourglass]Left: 0m 0s + (1m 20s) - this last part would increase until job ends
[image]

jordanruthe, 2021-01-21:
The latest update has the ability to change the print time estimate method. Please check that out and see if that fixes it for you.

hunagyp, 2021-01-25:
Thank you very much @jordanruthe , it looks good to me with these options.

closed 2021-01-26 by jordanruthe