KlipperScreen/KlipperScreen #54: BED_SCREWS_ADJUST macro support

cretells, 2021-01-15:
Currently the menu for Z Bed Leveling seems to manually find the screws within the config file and manually moves to each  position. 

I have had a problem with the lower left on my Ender 5 Plus where the location that the menu sends the head to is ~.3mm lower than it should be.

Would it not be a simpler approach to rely on the Klipper FW command BED_SCREWS_ADJUST similar to what is done for PROBE_CALIBRATE? This would allow a simple interface which could show the location of the head as BED_SCREWS_ADJUST moves around the plate and the ACCEPT / ABORT / ADJUST buttons on the menu

jordanruthe, 2021-01-26:
I don't know that I would want to force the BED_SCREWS_ADJUST command on that panel, as it's nice to be able to jump to any location. That being said, I'm not opposed to adding some kind of functionality in for that. 

For the problem of it going to ~.3mm lower than it should be, the actual gcode being run is:
```
G1 Z7 F800 # Go to a height that shouldn't hurt the bed
G1 X* Y* F3600 # Go to the position of the screw
G1 Z.1 F300 # Lower to just above bed
```

There might be a homing problem as Z.1 should be above your bed slightly, not in your bed.

jordanruthe, 2021-02-19:
I haven't heard a response to this ticket. If you are still having an issue, please re-open this.

closed 2021-02-19 by jordanruthe

cretells, 2021-02-19:
Sorry for the lack of response. I played with this a little more it appears to be fine functionality and as you say better than the automated script since you can home to each location.