KlipperScreen/KlipperScreen #55: Difficulty with install scripts

rcfishhunt, 2021-01-15:
First of all I would love to say that your work looks amaizing for the raspi gurus that can get it installed and operating.  If its not too much to ask,  Can you make these instructions a little more novice friendly?  I want to use it on my Voron badly but I just can't get it to boot up automaticaly.  

Last command showing up on my pi is 6.557421] systemd[1]: listening on udev Kernel Socket

pi@mainsailos:~ $ ls
CanSerial    klipper_config  LCD-show       moonraker
gcode_files  KlipperScreen   mainsail       moonraker-env
klipper      klippy-env      mjpg-streamer
pi@mainsailos:~ $

It is installed

jordanruthe, 2021-01-19:
I believe you got this running based on conversations in the discord channel. Do you still need help?

rcfishhunt, 2021-01-19:
Yes thank you!   I forgot to check back here too.   One thing though, if
anyone else new comes along simplified install instructions would help
alot.   My biggest problem was following the instructions and came across
the create virtual environment.   I had no idea what that was.

I copy and pasted the entire install script and ran in sudo su to prevent
having to put my password in mid script.   I think it was missing one file
that it was entering as my password.

Thanks for all you do!

Seems great so far. I will be following your development!

Greg

closed 2021-03-31 by jordanruthe