KlipperScreen/KlipperScreen #57: KlipperScreen says it encountered an error

nicedevil007, 2021-01-17:
Heya,

I updated to latest version and it seems after Homing all axis KlipperScreen stops to work with this message:

[image]

Going to the WebUI everything is working and printer is still printing...
[image]

jordanruthe, 2021-01-17:
Can you post the KlipperScreen and Moonraker log files from /tmp?

nicedevil007, 2021-01-18:
yesterday it didn't happen again, don't know why. Tomorrow it happened again around 6:15-6:20.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5828038/KlipperScreen.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5828039/moonraker.log)

nicedevil007, 2021-01-18:
Is there a possible way to remove KlipperScreen. I installed it with root and everything else was installed with the pi user with the KIAUH script compilation. So maybe I have to install your software with KIAUH as well? So everything is installed with the pi user.

jordanruthe, 2021-01-19:
Hm, I'm not seeing why it would change to an error state from the log files you attached.

To remove KlipperScreen, run `sudo rm /etc/systemd/system/KlipperScreen.service` and delete the directory your originally installed it as.

Frostie2k, 2021-01-19:
Same problem here, when i home the axis or start a print on the WebUI Klipper screen stops working.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5837618/KlipperScreen.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5837619/moonraker.log)

jordanruthe, 2021-01-19:
I just pushed an update that will enable more debugging on the state issue. Would you be able to update and try again with the latest and, when the problem happens, upload a new log file?

NBouquain, 2021-01-19:
I have just pulled the new commits in hope that the issue was fixed since for me too KlipperScreen spuriously reset to the Shutdown screen each time a homing command was send (either from KlipperScreen or the printer screen or Fluidd) and remains stuck in this state.
The issue was introduced with the state transition changes c58b040

With the modification in fc074c3 the behavior is even worse as now it does not switch to the main panel when the printer is turned on.
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5838360/moonraker.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5838362/KlipperScreen.log)

nicedevil007, 2021-01-19:
> I just pushed an update that will enable more debugging on the state issue. Would you be able to update and try again with the latest and, when the problem happens, upload a new log file?

I installed latest version now  (still with root user instead of pi). Rebooted the pi.... problem not there anymore o_O

Before I could do reboot of the Pi or Klipper/Fluidd as much as I can... always showing the error page.
Mine wasn't crashing on the WebUI (Fluidd), just KlipperScreen.

So maybe I'm lucky right now or something else changed?

jordanruthe, 2021-01-20:
@NBouquain From those log files, it looks like moonraker was sending notification that Klippy has shut down. Can you check the Klippy log file to see if an error was causing it to shutdown? KlipperScreen was just reacting to the shutdown notification.

jordanruthe, 2021-01-20:
@Frostie2k I'm also unable to tell what exactly is happening from your log files. I've added some additional debugs, can you please update to the latest commit and re-post log files after the issue?

NBouquain, 2021-01-20:
@jordanruthe It may comes from the way my printer is connected.
I have a tplink smartplug managed by Moonraker that control the mainboard power while the raspberry is powered separately and remains powered while the printer is 'off'.
Consequently, when the printer is 'off', Klipper enters shutdown state as it loose communication with the MCU. When the printer is turned 'on' through Moonraker api, a udev rule triggers a Klipper restart (once the USB connection is detected). Hence during the restart Moonraker may send the notification that Klipper has disconnected.

The good news is that with lasted commit a51360f my previous issues (spurious transition to shutdown or remaining stuck in shutdown screen) has been fixed. The shutdown screen with message 'Klipper has shutdown' is displayed when the printer is 'off', then the message 'Klipper has disconnected' during restart and then the main screen.

Test was performed with lastest version of Moonraker (v0.2.8-32) / Klipper (v0.9.1-161) / Fluidd (1.5.1)

[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5845297/moonraker.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5845299/KlipperScreen.log)

Frostie2k, 2021-01-21:
@jordanruthe 
the problem no longer occurs on my printer since I installed the update.

closed 2021-02-19 by jordanruthe