KlipperScreen/KlipperScreen #58: Looses connection when a print job is started by another moonrake client

Nandox7, 2021-01-20:
Hi,

Reporting this isue here that I'm seeing.
I have Klipperscreen running alongside FluidDD UI, in case I start a print using Fluidd, KlipperScreen loose the connection.

IT says that Klipper shutdown (that is not true) and if I force restart the KlipperScreen service It stays forever with the "initializing message in the LCD.
Checking the log I can clearly see that it was able to comunicate an it even received the status that klipper is printing.

```
ccel_to_decel': 7000.0, 'axis_minimum': [0.0, 0.0, -0.5, 0.0], 'axis_maximum': [260.0, 240.0, 250.0, 0.0], 'position': [112.983, 115.928, 0.019159361882638067, 89.45488000000044], 'extruder': 'extruder'}, 'fan': {'speed': 0.0}, 'webhooks': {'state': 'ready', 'state_message': 'Printer is ready'}, 'heater_bed': {'temperature': 86.00248219247557, 'power': 0.1696788422897857, 'target': 85.0}, 'extruder': {'pressure_advance': 0.1275, 'target': 240.0, 'smooth_time': 0.04, 'power': 0.3150097760102062, 'temperature': 240.1732832036485}}
2021-01-19 20:35:30,123 - KlipperScreen.Printer - DEBUG - Changing state from 'disconnected' to 'printing'
2021-01-19 20:35:30,123 - KlipperScreen.Printer - DEBUG - Running callback for state: printing
2021-01-19 20:35:30,483 - KlipperScreen.KlipperWebsocket - DEBUG - Attempting to reconnect
2021-01-19 20:35:30,484 - KlipperScreen.KlipperWebsocket - DEBUG - Reconnected
2021-01-19 22:11:46,596 - KlipperScreen - DEBUG - [
  "notify_gcode_response",
  "Done printing file"
]
````

Later also when the print is finished, yet it does not do anything.

jordanruthe, 2021-01-20:
Can you post the complete KlipperScreen log file? I've also added some additional logging statements in the latest commit which would help me track down issues like this.

jordanruthe, 2021-01-23:
Ok, I'll leave this open for a few days longer. If you run into it anymore, let me know.

Nandox7, 2021-01-25:
Thanks.
I'll go ahead and set this as closed, For the past two days I haven't seen one instance of this happening again so I'll assume this is no longer a problem.

closed 2021-01-25 by Nandox7