KlipperScreen/KlipperScreen #59: Any g-code or macro disconnects Klipperscreen

Mosher23, 2021-01-20:
Hello,

so i recently installed KlipperScreen on my Voron and run into an issue.
After i start or restart Klipper Klipperscreen boots ok into the main screen. BUT after i run any command (g-code or macro) it disconnects from Klipper and shows me on-screen message "Klipper has shutdown". Even it further runs without any issue on Mainsail.

I run following firmware: 
- Klipper v0.9.1-161
- Moonraker v0.2.8-32
- Mainsail v1.0.0

jordanruthe, 2021-01-20:
Can you post /tmp/KlipperScreen.log and the moonraker.log file?

Mosher23, 2021-01-20:
Yes, sure, here
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5844269/KlipperScreen.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5844270/moonraker.log)

jordanruthe, 2021-01-20:
I'm having a hard time telling exactly what's going on here. I've included some extra logging with the last couple of updates. Would you be able to update to the latest commit, re-try the macros, and post the new logs?

Mosher23, 2021-01-20:
I am not familiar with update process for KlipperScreen. Do i just reinstall it?

jordanruthe, 2021-01-20:
This assumes your directory for installation is `/home/pi/KlipperScreen` and will work to update. Note: if you've modified any base files, this won't work.

```
cd /home/pi/KlipperScreen
git pull
sudo systemctl restart KlipperScreen
```

Mosher23, 2021-01-21:
So i have updated it!
Issue still persists
But getting other Message now "Klipper has encountered an error. Issue a FIRMWARE_RESTART..."

Here are the logs

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5847280/KlipperScreen.log)
[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5847281/moonraker.log)

Mosher23, 2021-01-21:
Issue was fixed with a very recent update!

closed 2021-01-21 by Mosher23