KlipperScreen/KlipperScreen #60: Can get it to start

michelluccote, 2021-01-21:
I've getting KlipperScreen to run well on one of my printer. I'm trying to setup a second printer with a Raspberr PI 3B+. I did the install and I can run the "sudo service KlipperScreen start" command but nothing happens. The log file is not created so I don't have any visibility into what the service is doing when it tries to start. Where can I look to find any errors encountered during service startup?

michelluccote, 2021-01-21:
I found the service log file file. I believe I have encountered this before in my first install but cannot recall what was wrong with the setup to make it work. Here is the service log file content.

Jan 21 04:51:43 raspberrypi systemd[1]: Started KlipperScreen.
Jan 21 04:51:43 raspberrypi systemd[1]: Stopping KlipperScreen...
Jan 21 04:51:43 raspberrypi xinit[4610]: XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
Jan 21 04:51:43 raspberrypi xinit[4610]: /usr/bin/xinit:       after 7 requests (7 known processed) with 0 events remaining.
Jan 21 04:51:43 raspberrypi xinit[4610]: unexpected signal 15
Jan 21 04:51:43 raspberrypi systemd[1]: KlipperScreen.service: Main process exited, code=exited, status=1/FAILURE
Jan 21 04:51:43 raspberrypi systemd[1]: KlipperScreen.service: Failed with result 'exit-code'.
Jan 21 04:51:43 raspberrypi systemd[1]: Stopped KlipperScreen.
Jan 21 04:51:43 raspberrypi systemd[1]: Started KlipperScreen.
Jan 21 04:51:44 raspberrypi xinit[4617]: (EE)
Jan 21 04:51:44 raspberrypi xinit[4617]: Fatal server error:
Jan 21 04:51:44 raspberrypi xinit[4617]: (EE) Server is already active for display 0
Jan 21 04:51:44 raspberrypi xinit[4617]: #011If this server is no longer running, remove /tmp/.X0-lock
Jan 21 04:51:44 raspberrypi xinit[4617]: #011and start again.
Jan 21 04:51:44 raspberrypi xinit[4617]: (EE)
Jan 21 04:51:44 raspberrypi xinit[4617]: (EE)
Jan 21 04:51:44 raspberrypi xinit[4617]: Please consult the The X.Org Foundation support
Jan 21 04:51:44 raspberrypi xinit[4617]: #011 at http://wiki.x.org
Jan 21 04:51:44 raspberrypi xinit[4617]:  for help.
Jan 21 04:51:44 raspberrypi xinit[4617]: (EE)
Jan 21 04:51:59 raspberrypi xinit[4617]: XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
Jan 21 04:51:59 raspberrypi xinit[4617]:       after 7 requests (7 known processed) with 0 events remaining.
Jan 21 04:51:59 raspberrypi systemd[1]: KlipperScreen.service: Succeeded.

michelluccote, 2021-01-21:
I was finally able to get it to run. The installation does not seem to work by default on a standard raspbian installation. I had to install a number of other packages. I am not entirely sure which of these causes it to run but I installed these:

	sudo apt-get install --no-install-recommends xserver-xorg
	sudo apt-get install --no-install-recommends xinit
	sudo apt-get install lxde-core lxappearance
        sudo apt-get install lightdm
	
I also did some commands to give permission to the pi account to control tty. I am still not clear which option allowed this to run but the installation is still not straight forward.

I will also add that using kiauh by default did not work. I installed everything but I got errors installing KlipperScreen. I had to do a manual install to fix the issue. I will likely create a few more installed in the coming weeks and next time I'll take it step by step to try and understand what is really missing in the installation process to make this completely smooth.

jordanruthe, 2021-01-21:
Are you using the raspbian lite image? If you're using a regular raspbian image, there may be issues with lightdm being installed.

michelluccote, 2021-01-21:
I see. I may have missed that detail. I installed the regular raspbian image so that maybe the problem.

michelluccote, 2021-01-22:
I'll close the issue for now and for my next install will go with raspbian lite. If I still have issues then I can create a new issue.

closed 2021-01-22 by michelluccote

SXRguyinMA, 2023-01-14:
did it work with the lite version? I can't get klipper screen to work for the life of me on my 3b+ no matter what I do