KlipperScreen/KlipperScreen #61: Very small text and images on 480x320 screen

michelluccote, 2021-01-21:
I'm setting up klipperscreen on a 480x320 screen. It works and all buttons seem operational but the icons and text is very small even if the screen seems to have some space to spare. It would be great to take better advantage of the space available and make the icons and text a bit larger.

jordanruthe, 2021-01-23:
I'll look at implementing a settings options to increase font/image size in the near future. However, it may provide unexpected results as the width of elements can be determined by the longest phrase (I think "Disable Motors" might be the longest menu phrase) and then could possibly scale buttons/items off the screen.

Sergey1560, 2021-01-27:
I also use a 480x320 screen and it would be great to make bigger icons. Especially the back button, it's hard to hit.

michelluccote, 2021-01-27:
If string length is an issue it would be valuable to allow scaling of text independently of scaling of buttons. This way we can set optimal numbers for each one to get the largest buttons possible and a smaller increase in text size that would not causing the text to make the UI go off screen. 

Thank you as always for your prompt response.

sudomofo, 2021-02-15:
can you at least move the back button to the center a bit. really hard to press on 3.5" screen. good job btw

michelluccote, 2021-02-17:
You could move it out a bit or make it a bit larger. When installing the screen in a printed case it can be hard to reach the button if the case top is just a bit too thick. I redesigned my case to have a bevel so I can more easily access it and that worked but a small tweak would help as well. Thanks for the great work on this UI. I use it on multiple printers and find it very functional.

jordanruthe, 2021-02-19:
With the latest updates, the design has been changed for the back/home/estop buttons. Please let me know if this works better.

michelluccote, 2021-02-19:
I like it. I wasn't sure at first but I got use to it really quickly. I'll close this. I'm happy and fully functional at this point.

closed 2021-02-19 by michelluccote