KlipperScreen/KlipperScreen #62: Not switching to printing screen

wormiedk, 2021-01-23:
I have had this work only a few times but recently when I upload files and set prusaslicer to "start print after upload" klipper screen does not switch to the printing view. It still display temperature and such but does not switch to the printing view.

If I kill the klipperscreen process it is stuck at "initializing". If I start a print directly from the display it also does not show the printing screen.

I have attached the log file
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5860339/KlipperScreen.log)

jordanruthe, 2021-01-23:
Can you run the following command? `journalctl -u KlipperScreen > ks.log` and upload the result ks.log file?

wormiedk, 2021-01-24:
Actually it seems reinstalling has solved the problem for now

davide-molino, 2021-01-28:
I've the same problem, but also when starting from the lcd or from fluidd. Reinstalling KlipperScreen doesn't solve the problem.
I attach the ks.log file:
[ks.log](https://github.com/jordanruthe/KlipperScreen/files/5889089/ks.log)

jordanruthe, 2021-01-29:
@davide-molino Are you using the filament method for the timing estimation? If so, I just fixed it in an update

davide-molino, 2021-01-30:
@jordanruthe no, initially I was using the file estimation method, but it wasn't very precise. Now I'm using the Slicer method. It worked after the second reinstallation, but now it didn't works again

jordanruthe, 2021-02-04:
@davide-molino Looking at your log, commit ad171313939c738b3b39a9c2f05eef9f702fb289 should fix this issue I see happening.

closed 2021-02-19 by jordanruthe