KlipperScreen/KlipperScreen #64: Should have a KlipperScreen restart button on the klipper disconnected screen

michelluccote, 2021-01-28:
One a few occasions I've had KlipperScreen disconnect from klipper and show the "Klipper has disconnected" screen. I need to send a log of this time since it never reconnected after I just did a Klipper restart from Fluidd. So I know my Klipper is running but KlipperScreen is stuck on this screen. My only options are Restart which I'm assuming means kipper restart and Firmware restart. 
I'm now stuck logging into my PI and restarting KlipperScreen or possibly restarting Moonraker. Or I can shut it down and restart the whole system.
This screen instead should have the extra option of restarting Moonraker or more importantly restarting KlipperScreen. This would just do the simple service stop and service start on KlipperScreen (or Moonraker) which would allow an easier recovery of my control screen.

jordanruthe, 2021-02-26:
Ideally, this wouldn't be needed. However, I'll probably add a restart button in the system panel in the near future.

jordanruthe, 2021-03-31:
This has been implemented in the system panel (which is accessible from the disconnected screen)

closed 2021-03-31 by jordanruthe