KlipperScreen/KlipperScreen #67: [Enhancement] Button for Webcam image

nicedevil007, 2021-01-30:
Hey again :)

just an idea, dont know if that can be done or not :/

A button to open livewebcam view with a setting to enter the right IP/port config for the cam.
This can be very handy on setup the printer. That is the only thing missing in my opinion to get everything working with "just" KlipperScreen.

Everytime I have to take my phone or laptop to see if the camera is correct algined or not.

Not a huge deal, but would be reaaaaaaly nice.

jordanruthe, 2021-02-04:
Thanks for this use case, I'll look into this at some point in the future.

schulzh, 2021-11-07:
This would be great for camera angles you cannot view from outside the printer, especially nozzle cams like this: https://www.youtube.com/watch?v=GAp23w_dnNc

Martyfree123, 2022-06-26:
Any update on this? I'm hoping for the same thing.

RugerLuger, 2022-06-30:
I would also love to be able to view webcam from my klipperscreen.  Mt printer is fully enclosed with opaque plastic and a webcam is the only way i can see in. It would be soooo nice to toggle that view on the touchscreen vs pulling it up on my pc.

goldjunge91, 2022-08-31:
me too to have on my touch panel a webcam button or menu point that i can see my webcams for my printer i have two cams one nozzle one chamber cam.

adrianocr, 2022-09-10:
I too think this would be pretty cool. I'm running klipperscreen on an android phone so there's a nice screen for it and everything.

commit d2d2c820 (2022-12-11): feat: camera support, using mpv as backend

it should be able to play any video or stream that mpv itself can handle

close #67 close #129
closed 2022-12-11 by alfrix