KlipperScreen/KlipperScreen #70: Fluiddpi

pannik72, 2021-02-01:
Good day

Thanks for amazing job.
I have tried today to install KlipperScreen on FluiddPi fresh installation and installation finished successfully.
However the KlipperScreen can not communicate with klipper.
It has to be noted that printer.cfg on this distro is not in the pi home folder as in octoprint but in another folder.
I don't know if this the problem, but I am a novice and I don't know what to to in order to resolve the issue.

Thanks and best regards

jordanruthe, 2021-02-04:
KlipperScreen doesn't care about the klipper configuration location. As long as KlipperScreen can talk to moonraker it should work.

jordanruthe, 2021-02-19:
If you're still having issues, please re-open this issue.

closed 2021-02-19 by jordanruthe