KlipperScreen/KlipperScreen #71: KlipperScreen stopped showing print screen

Potplanty, 2021-02-02:
Not sure what caused it (I'm assuming the latest repo update) but the screen doesn't actually show job progress any more. You can initiate the job from KS and everything else works but the actual job screen doesn't show up at all.

It also appears a couple of the menus are out of alignment now too. For example the bed mesh screen looks completely wrong to what you usually see.

klipper v0.9.1-189
KlipperScreen v0.1.3-37
moonraker v0.2.8-52 

[moonraker.log](https://github.com/jordanruthe/KlipperScreen/files/5910365/moonraker.log)
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/5910366/klippy.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5910367/KlipperScreen.log)

Potplanty, 2021-02-03:
Alright, somehow without any changes it appears the print screen is back. Still appears the bottom of the screen is cut off though as I can't see the coloured lines underneath the buttons.

jordanruthe, 2021-02-04:
Commit b258d64fb44f8d1ce4859b33b4a75c43b485282b should fix the buttons being cutoff.

Potplanty, 2021-02-04:
> 
> 
> Commit [b258d64](https://github.com/jordanruthe/KlipperScreen/commit/b258d64fb44f8d1ce4859b33b4a75c43b485282b) should fix the buttons being cutoff.

Working great, thanks.

Potplanty, 2021-02-04:
Just realised the print status screen has stopped showing again. Can initiate prints but nothing shows on KlipperScreen regarding the current status, just sits on the main menu.

jordanruthe, 2021-02-04:
Can you upload the KlipperScreen.log file?

Potplanty, 2021-02-04:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/5927992/KlipperScreen.log)

jordanruthe, 2021-02-05:
Commit 31ae88596c0ed0f7bb95162edeca6cacb1f82e4c should fix this problem.

Potplanty, 2021-02-06:
> 
> 
> Commit [31ae885](https://github.com/jordanruthe/KlipperScreen/commit/31ae88596c0ed0f7bb95162edeca6cacb1f82e4c) should fix this problem.

That's got it. Thanks mate!

closed 2021-02-06 by Potplanty