KlipperScreen/KlipperScreen #72: [Bug] Printscreen => Buttons a little bit outside the screen

nicedevil007, 2021-02-03:
Heya,

I have installed latest version of KlipperScreen and today I found a "cosmetical bug". The buttons are too low. I guess that happens on too long filenames for gcodes. Maybe you can "cap" the maximum number of characters for the filename that is shown in the printscreen.

Maybe it isn't usefull to show the fileextension because it is always ".gcode" :D So we save some charackters here.

All other menus are working fine, so it shouldn't be screen related issue.

[image]

jordanruthe, 2021-02-04:
Looks like the word wrapping in buttons is messing it up. The filename has another line it can spill over to still and shouldn't affect the spacing. Commit b258d64fb44f8d1ce4859b33b4a75c43b485282b should fix it

nicedevil007, 2021-02-04:
> Looks like the word wrapping in buttons is messing it up. The filename has another line it can spill over to still and shouldn't affect the spacing. Commit [b258d64](https://github.com/jordanruthe/KlipperScreen/commit/b258d64fb44f8d1ce4859b33b4a75c43b485282b) should fix it

is fixed <3

closed 2021-02-04 by nicedevil007