KlipperScreen/KlipperScreen #75: UpdateManager problem

nicedevil007, 2021-02-06:
Heya @jordanruthe,

just added your update possibility to moonraker and it is not working. Maybe something changed in the last fixes/commits?

That is my snippet of the moonraker.conf

```
[update_manager]
[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd

[update_manager client KlipperScreen]
type: git_repo
path: /home/pi/KlipperScreen
origin: https://github.com/jordanruthe/KlipperScreen.git
env: /home/pi/.KlipperScreen-env/bin/python
requirements: scripts/KlipperScreen-requirements.txt
install_script: scripts/KlipperScreen-install.sh
```

And this is  what is shown to me in the webui:

[image]

nicedevil007, 2021-02-07:
[image]

closed 2021-02-07 by nicedevil007

Elektrickser, 2021-02-08:
How you solved? I have the same problem.

nicedevil007, 2021-02-08:
> How you solved? I have the same problem.

Click on the link of the other referenced post ;) I wrote everything there.

Elektrickser, 2021-02-08:
Thanks, I was blind.