KlipperScreen/KlipperScreen #76: [Enhancement] When multiple options are presented, highlight the currently selected one.

Potplanty, 2021-02-09:
Just a thought that I feel is (probably) easy to implement and would make things a lot clearer. When on a screen with multiple selection options (eg, temperature screen, movement screen) have it so that the current option is highlighted in some way to show it's the current option. Whether that be making the selected icon inverse colour or putting a red ring around it or something.

jordanruthe, 2021-02-14:
For the multiple options, are you referring to the temperature/distance numbers? Such as 5, 10, 25C for temp?

Potplanty, 2021-02-14:
> For the multiple options, are you referring to the temperature/distance numbers? Such as 5, 10, 25C for temp?

Yes that's exactly what I'm referring to.

jordanruthe, 2021-02-14:
The currently selected option has a different color background. For instance, 1 and Medium are selected in this image: https://github.com/jordanruthe/KlipperScreen/blob/master/docs/img/extrude.png

Potplanty, 2021-02-15:
Have literally never seen that on all my installs. On the exact same screen I have nothing of the sort, only the one colour for all buttons. Can see that colour if I hold my finger on a button but it goes away as soon as I let go.

closed 2021-02-20 by Potplanty

jordanruthe, 2021-02-20:
If you're still having issues with this, can you post a picture along with your KlipperScreen.log?