KlipperScreen/KlipperScreen #80: Klipper screen is not going to sleep

bakaufman, 2021-02-13:
I have tried a number of sleep settings and cannot get the screen to go to sleep. Is there something with config, lcd-show driver, raspi-config that needs to change?
Thanks!

jordanruthe, 2021-02-26:
One update I just pushed changes screen blanking to being able to use screen power save. Let me know if that has any effect. Output from the command `xset q` would be helpful if you're still having issues. You'll have to run that command after KlipperScreen has started.

grigi, 2021-02-26:
Hi I have it now where my display goes to sleep, but if you touch the display to "turn it on", it will "click" where you press.
This is causing some unwanted actions.

Is there a way you could determine if the display has been blanked?
A quick check found that `xset q` will report "Monitor is in Standby" when monitor is blanked and "Monitor is On" when awake.

jordanruthe, 2021-03-31:
The first click after the display goes to sleep should now be ignored. If you're still having issues, we can reopen this.

closed 2021-03-31 by jordanruthe