KlipperScreen/KlipperScreen #81: [Enhancement] Add Setting to adjust Screen brightness

nicedevil007, 2021-02-13:
heya @jordanruthe 

again, I found a small thingi that would be nice to be added to your KlipperScreen.

Adjusting the brightness (tested with RaspberyPi 7" Touchscreen) is working with this command:

`sudo sh -c 'echo "128" > /sys/class/backlight/rpi_backlight/brightness'`

the number `128` can be anything between `0` and `255`.

Zayl1up, 2021-02-13:
it would be awesome to see this feature in the feature!
Right now I'm using the console. The black looked a bit too much grey :)

jordanruthe, 2021-02-14:
Thanks for this writeup. This shouldn't be too hard to address and I'll be looking at it in the near future

emtrax-ltd, 2022-12-10:
Here are my thoughts on configuring this function:

1) Configure a specific command (script, binary etc.) with a placeholder (parameter) for the value as to be set.
2) Configure the minimum and maximum values ​​that can be passed to the command.
3) Interpolate/calculate the ui value "0-100%" based on the minimum and maximum values.

For many waveshare displays it is possible to control the backlight via PWM -> "gpio -g pwm 18 $value" (control brightness, $value is between 0~1024)

Peuqui, 2023-11-23:
I found out recently, that waveshare displays are able to dim/shutoff the backlight with PWM:
<
gpio -g pwm 12 1024
gpio -g mode 12 pwm #set the pin as PWM
gpio pwmc 1000
gpio -g pwm 12 X #change the brightness, X ranges 0-1024
>

It would be nice, to shutoff the backlight when KlipperScreen shuts off the display, too.
This needs a customization option due to different displays attached to the various raspis.
Further, the option to dim the the backlight with defined gcode macros is nice too.

alfrix, 2023-11-24:
There is also another alternative, that i described a while ago on discord

you can create a [moonraker power device that uses gpio](https://moonraker.readthedocs.io/en/latest/configuration/#gpio-device-configuration) or a [klipper_device that uses a macro or pin](https://moonraker.readthedocs.io/en/latest/configuration/#klipper-device-configuration), and then configure the options `screen_on_devices` and `screen_off_devices` of [klipperscreen](https://klipperscreen.readthedocs.io/en/latest/Configuration/#main-options) to trigger said power device, the only drawback is that moonraker needs to be connected

Peuqui, 2023-11-24:
Brilliant! I'll try it out.

LazaroFilm, 2025-04-02:
That gave me an idea for the BTT HDMI5 which only has physical buttons on the edge to control brightness as far as I know. But I may put two small mosfets wired and soldered the switches pins to a GPIO pin. As pressing one all the time sets the screen at full brightness and the other gets it completely off. I could at least achieve full brightness or completely off.