KlipperScreen/KlipperScreen #83: Color for Powerplug on "deactivated" State is different than on other menus

nicedevil007, 2021-02-14:
Just small cosmetic thing.

The color for the font is different during a print of the power plugin in comparison to services f.e.

It is much darker:
[image]

Here the services in "light grey":
[image]

jordanruthe, 2021-02-14:
This looks to be a fluidd issue, not KlipperScreen. You should repost this on the fluidd github issues.

nicedevil007, 2021-02-14:
> This looks to be a fluidd issue, not KlipperScreen. You should repost this on the fluidd github issues. 

Haha lol! Sry, was sure I was on the Right repo to post 😇😂😂😂

closed 2021-02-16 by jordanruthe