KlipperScreen/KlipperScreen #87: After reboot Border around Extruder0 temp

nicedevil007, 2021-02-21:
I recently installed fluiddpi from scratch and then installed KlipperScreen from scratch as well.

After last updates on KlipperScreen a reordering of the Ui elements was done. So here and there a border was added and removed again but the border around the Extruder0 temp (white at top and dark grey around the other edges) is still present after a reboot of the pi.

After touching on the extruder0 one time the border disappears. 100% reproduceable to me.

I know just small cosmetic thing :)

jordanruthe, 2021-02-22:
c8fe6de747e44d2f3d0d303abda4a5899d8ab8db fixes this

jordanruthe, 2021-02-22:
Forgot to add, there's an external dependency, so you'll have to re-run the installation script.

nicedevil007, 2021-02-22:
So I reinstalled right now with your latest commits, I guess the installationsscript was done for that right?

[image]

still there after reboot.

jordanruthe, 2021-02-22:
Did you re-run the `KlipperScreen-install.sh` script after pulling the update? Essentially I'm running a command to move the mouse location, which you can run yourself to verify it's installed/working. `xdotool mousemove 0 0`

nicedevil007, 2021-02-22:
I reinstalled it with Fluidd and its shown possibility to update KlipperScreen via Moonraker API.
Can test after print again :)

nicedevil007, 2021-02-22:
ok, seems to be fixed :) thank you!

closed 2021-02-22 by nicedevil007