KlipperScreen/KlipperScreen #88: [Enhancement] KlipperScreen.conf also be able to put into klipper_config

nicedevil007, 2021-02-21:
I'm sure everybody is using any king of webui for controlling klipper.
We are able to see and edit all those config files to control klipper or moonraker...etc... that are located inside the `klipper_config` folder of the pi users home directory.

So maybe you can add the possiblity to not just look for the `KlipperScreen.conf` inside `/home/pi` directory, instead give us the possiblity to let this conf file beeing located inside `/home/pi/klipper_config`

Usecase: we can then edit this file from the Webui as we do for all other configs.

Raabi91, 2021-02-21:
you can do a simple symlink to yout config

Elektrickser, 2021-02-22:
make the follow comands on your system:
 cd /home/pi/klipper_config/
ln -s "/home/pi/KlipperScreen.conf" KlipperScreen.conf

maybe @jordanruthe can take this in the installation routine.

jordanruthe, 2021-02-22:
It's possible to edit `/etc/systemd/system/KlipperScreen.service` to add an argument to the startup command and change the config file location. The config file argument is -c

Elektrickser, 2021-02-22:
Maybe its better with the symlink. Many users use the actual location. And with this feature you can help all users.

nicedevil007, 2021-02-22:
Or just add an option during the installation process of your script to ask the user for the configuration folder, if it exists, place a dummy config there and if it doesn't exist create the folder and place a dummy file inside there.

Dummy file because it isn't needed to change anything to get your awesome addon working.

For now I'm using the symlink, thanks for the tip :)

einencool, 2021-07-04:
Hello, Ive got one question.
I've added the symlink as mentioned above with putty, and it seems, that everything is fine, because I can see and edit the file from the Mainsail UI.
But the changes I make are only in the "Symlink" but it seems, that it's not a link, it's a file on its own.
The changes are not mirrored to the /home/pi/KlipperScreen.conf file.

I've tested it on two printers, but the behaiviour is the same...
Can someone help me in this case? I'm not an linux expert and maybe I've done something wrong.

alfrix, 2021-09-11:
This should be fixed and it can be closed unless i missed something

closed 2021-09-20 by jordanruthe