KlipperScreen/KlipperScreen #89: [Enhancement] Filament Change procedure

nicedevil007, 2021-02-22:
Heya... sry for posting that much first :/
But hope my ideas make your piece of software even better and better. Btw where is your donation link :D

So we all know how to get the filament change done inside a gcode on klipper but it is much more complicated (maybe just to me because I don't know how to do it right?) than it was on marlin.

So what I would like to see:

Icon for filament change. if we hit the button a new window pops up where we can enter the filamenttemperature needed for the change and a button to start the procedure.
For now I have a macro that always heats up to 225°C and that can be bit too low for ABS f.e.
If I throw this part of the macro out I have to first go manually to different menu heat up the nozzle to the wished temperature and wait until this is reached and then start my gcode macro to change the filament.

So this would be a quality of life change to make things a lot easier for dayli use.

stenc55, 2021-03-15:
Agree. One of the things I miss since moving to Fluidd and Klipperscreen from Octo/ Octodash is dedicated filament change procedure as mentioned above. Except, to continue this procedure, button "Next" after inserting new filament and then extruder would push desired lenght of filament as stated in config menu. (filament path from gears to nozzle is different for every printer"

jordanruthe, 2021-03-15:
What if there was a filament load/unload button on the filament panel that mapped to macros for `LOAD_FILAMENT` and `UNLOAD_FILAMENT` where the user can define the lengths in the gcode macro?

stenc55, 2021-03-16:
Yes. That would be nice. Can it does two speeds, first fast to load and then slow to purge old filament from nozzle?
If it is of any help, those two I am using now. I have Hemera so path is very short.
[gcode_macro Filament_change_start]
gcode:
 G91
 M109 S200
 G1 E-30 F600

[gcode_macro Filament_change_end]
gcode:
 G91
 G1 E25 F600
 G1 E30 F50
 M104 S0

photonlightcom, 2021-07-06:
> What if there was a filament load/unload button on the filament panel that mapped to macros for `LOAD_FILAMENT` and `UNLOAD_FILAMENT` where the user can define the lengths in the gcode macro?

Not sure what the status of this suggestion is, but has this been implemented yet?  Tying in to user-created load/unload macros sounds like a perfect solution since I already have the macros - would love some pretty buttons to trigger those macros.

nicedevil007, 2021-07-06:
It is not implemented yet. My hope was to get a close behavior to what it was/is on marlin during a filament change.

I have those macros as well but I would like to have a button to tell the printer to load filament now and not to let it trigger after the runout sensor was activated again. So a bit guidance on the screen will be realy nice especially for newer users.

photonlightcom, 2021-07-15:
Understood.  That sounds a lot more involved (and more of a complete feature) than what I was envisioning for simple manual filament changes.  The macro menu will get me by just fine for now.

PR #250 (2021-09-17): Add unload/load and reorganize extrude panel
- [x] Use load_filament and unload_filament macros and hide the buttons if not present

Fix #89
This PR (a 4th extruder will replace the temp shortcut)
[image]

master
[image]

this should help with #245
closed 2021-09-17 by jordanruthe