KlipperScreen/KlipperScreen #90: Not pulling in config file

JaredC01, 2021-02-23:
Hi,

It seems I'm having an issue with a custom config file.  No matter where I put the file (tried both of the locations defined in the config section), I never get the custom config applied (just trying to use a custom temperature for a preheat).

Currently on the latest version, installed via the CLI commands, updated via Mainsail.  I can modify the main config file, but that also breaks Git.

jordanruthe, 2021-02-24:
Can you post your KlipperScreen.log?

JaredC01, 2021-02-25:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6045658/KlipperScreen.log)

JaredC01, 2021-02-25:
The config file is currently in ~/ (Home for the pi user, which is who owns the KlipperScreen directory).

jordanruthe, 2021-02-25:
You ran the installation script as root, so it is looking for the root user instead of your user. Either modify the user in /etc/systemd/system/KlipperScreen.service or re-run the installation script with running it as root.

JaredC01, 2021-02-25:
Interesting, I'm about 90% sure I did NOT run it as root, but I'll give it a reinstall and try again.  Thanks!

JaredC01, 2021-02-25:
It was indeed running as root.  Updated the service and testing again.

JaredC01, 2021-02-25:
Okay, now I realize WHY it was running as root.  Startx won't run as user "pi" even when granting permissions to do so, so I just ran it as root to avoid the issue.  That's entirely my fault, and my apologies for the trouble!

closed 2021-03-31 by jordanruthe