KlipperScreen/KlipperScreen #91: KlipperScreen Can't control heater

michelluccote, 2021-02-23:
I've had a few times when temperature control no longer works on my setup.  Looking at KlipperScreen I find this error message but on restart it starts working again. I've included a log when it stopped working and the same log includes when it started working again at the end.

2021-02-23 23:18:56,805 - KlipperScreen.TemperaturePanel - DEBUG - Dev stats extruder: {'temperature': 21.264802030784722, 'target': 175.0, 'pressure_advance': 0.06, 'smooth_time': 0.04, 'power': 0.0}
2021-02-23 23:18:56,813 - KlipperScreen - ERROR - Uncaught exception <class 'NameError'>: name 'heater' is not defined
Traceback:   File "/home/pi/KlipperScreen/panels/temperature.py", line 228, in change_target_temp
    self._screen._ws.klippy.set_tool_temp(self._printer.get_tool_number(heater), target)
NoneType: None
[klippy-2.log](https://github.com/jordanruthe/KlipperScreen/files/6032362/klippy-2.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6032358/KlipperScreen.log)
[moonraker-3.log](https://github.com/jordanruthe/KlipperScreen/files/6032361/moonraker-3.log)

jordanruthe, 2021-02-24:
There's two updates that occurred two days ago that fix this issue.

michelluccote, 2021-02-25:
I've just done my update and will check it later today. I will close the issue if it does not re-occur.

michelluccote, 2021-03-03:
Problem has not returned so all is good.

closed 2021-03-03 by michelluccote