KlipperScreen/KlipperScreen #92: error: [Errno 2] No such file or directory: 'vext': 'vext' during installation

michelluccote, 2021-02-24:
I use Kiauh to install klipper screen on multiple PI machines. When do the installation I always get this error.

Building wheels for collected packages: vext.gi
  Building wheel for vext.gi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/setup.py'"'"'; __file__='"'"'/tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-__gaowg8
       cwd: /tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/
  Complete output (29 lines):
  running bdist_wheel
  running build
  installing to build/bdist.linux-armv7l/wheel
  running install
  vext.gi Install
  running install_data
  creating build
  creating build/bdist.linux-armv7l
  creating build/bdist.linux-armv7l/wheel
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext
  creating build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  copying gi.vext -> build/bdist.linux-armv7l/wheel/vext.gi-0.7.4.data/data/share/vext/specs
  running install_egg_info
  running egg_info
  writing vext.gi.egg-info/PKG-INFO
  writing dependency_links to vext.gi.egg-info/dependency_links.txt
  writing requirements to vext.gi.egg-info/requires.txt
  writing top-level names to vext.gi.egg-info/top_level.txt
  reading manifest file 'vext.gi.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.txt'
  writing manifest file 'vext.gi.egg-info/SOURCES.txt'
  Copying vext.gi.egg-info to build/bdist.linux-armv7l/wheel/vext.gi-0.7.4-py3.7.egg-info
  running install_scripts
  Install vext files:
  error: [Errno 2] No such file or directory: 'vext': 'vext'
  ----------------------------------------
  ERROR: Failed building wheel for vext.gi
  Running setup.py clean for vext.gi
Failed to build vext.gi
Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/setup.py'"'"'; __file__='"'"'/tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-x61gvv2q/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-x61gvv2q/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/setup.py'"'"'; __file__='"'"'/tmp/pip-install-14105ytb/vext-gi_7ec4a47da3714b35b1f27bc075dc0a70/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-x61gvv2q/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.

jordanruthe, 2021-02-26:
This is already on the todo list, but it's not high on my priority list to fix currently as the installation still works properly.

michelluccote, 2021-03-03:
Correct the installation still works fine. Not sure the value of this item then. Do you prefer if I close this then since you already have it planned?

closed 2021-07-18 by jordanruthe